platformio/platform-ststm32 #327: STM32 w/stlink incorrect upload address

LinoBarreca, 2019-12-20:
Hi. It seems like [this issue](https://community.platformio.org/t/jlink-uploading-to-incorrect-address/6085/3) is still present.
When using stlink to just upload the file it's flashed at the wrong address (0x0) corrupting the board.
https://github.com/LinoBarreca/Marlin/pull/7#issuecomment-566729164

When debuging it, instead, it gets uploaded from the correct address.

LinoBarreca, 2019-12-20:
line 280 lacks the default address.
board.get("upload.offset_address", "")

valeros, 2020-02-07:
Hi @LinoBarreca ! What board do you have issues with?

LinoBarreca, 2020-02-07:
Hi @valeros board is SKR Pro 1.1

valeros, 2020-02-07:
Looks like something similar to #343, but with a different offset. Could you please try adding the following line to your environment in `platformio.ini` file:
```
platform = ...
board = ...
board_upload.offset_address = 0x08008000
```

LinoBarreca, 2020-02-07:
@valeros yeah, overriding it works....but problem isn't "how to workaround it?"...but rather "why does platformio, when i start debugging, uploads the firmware to the correct address while when I 'upload' it doesn't?" both commands use openocd...

valeros, 2020-02-07:
@LinoBarreca The reason is that in debug mode PlatformIO uploads your firmware to the target using `load` command from GDB, but in case with `upload` command the firmware is uploaded as binary using OpenOCD directly. The binary image doesn't have information about memory addresses, so we need to specify it explicitly.

LinoBarreca, 2020-02-16:
@valeros then it should be disabled at all if no address is specified because, as is, it destroys users' boards. (or even better take the correct adresses automatically from ELF)

valeros, 2020-02-16:
@LinoBarreca By default we let OpenOCD to decide where to upload a binary image, if this behavior is not appropriate for some reason, it should be specified in a board manifest (as it done for the boards in this repository). We don't maintain the manifest for `SKR Pro 1.1` board, so we cannot be held responsible for its contents. In my opinion, you'd better open an issue the Marlin repository.  As for the dynamic detection of the offset address, it sounds like a good idea, at least for ARM-based platforms. You can open a feature request here or even better submit a PR with this feature.

mahmoodfathy, 2021-03-16:
@valeros  am having a similar issue with stm32f103C8T6 blue pill board , when i start in debug mode , it does everything correctly while the upload command doesn't seem to let it work properly i don't know which offset should i start with?