platformio/platform-ststm32 #331: Randomly rebuilding mbed library

HeathPetersen, 2019-12-31:
The mbed library is being rebuilt at random times WITHOUT ANY file changes.  It is easily reproducible as follows:

```
mkdir x; cd x
pio init --board nucleo_l432kc --project-option "framework=mbed"
cat >src/main.cpp <<END
#include <mbed.h>
int main() {}
END
pio run
```

After that, consecutive `pio run` commands may or may not recompile the mbed libraries.

There appear to be multiple mentions of this bug from other people.  For examples, see the following:
* [Always full rebuild allegedly after V4.0.0 - is my platformio.ini file deficient?](https://community.platformio.org/t/always-full-rebuild-allegedly-after-v4-0-0-is-my-platformio-ini-file-deficient/9327)
* [PIO/VSCode recompiles everything, every time](https://community.platformio.org/t/pio-vscode-recompiles-everything-every-time/10135)
* [Entire MBED recompiles every time](https://community.platformio.org/t/entire-mbed-recompiles-every-time/6766)
* [Randomly rebuilding all mbed libraries](https://community.platformio.org/t/randomly-rebuilding-all-mbed-libraries/9628/3)

Now, I admit that some of these may be something else.  However, the fact that multiple people are mentioning this, it only seems to be mentioned related to the mbed framework, and it is easily reproduced as shown above leads me to believe there is a genuine bug.

I am running platformio on Ubuntu 19.04 - fully upgraded.  The `platformio` / `pio` commands start with `#!/usr/bin/python3` so it should be using python 3.  `python3 --version` returns `Python 3.7.3`.  I have tested using platformio v4.1.0 and v4.1.1b5.

FYI - I did a very lengthy / detailed comparison of the whole project and .platformio directorys' timestamps and file sizes to see if something was changing without my knowledge.  I was unable to find anything changing outside of the generated files during runs.

ivankravets, 2019-12-31:
Thanks! I moved this issue to ST STM32 dev/platform. @valeros will take a look.

HeathPetersen, 2020-01-05:
I tried that fix earlier per the recommendation of @nmathon .  It does not appear to be the same problem.

HeathPetersen, 2020-02-03:
This is an extremely debilitating bug that significantly curtails productivity.  Is anyone actively working on this?  I am willing to provide any and all assistance in tracking it down.  I just don't have the pio / python experience to do it on my own.

ivankravets, 2020-02-04:
@HeathPetersen please contact @valeros  directly via https://community.platformio.org/u/valeros/summary , he will provide further help

mastrogippo, 2020-03-12:
> Hi, I published a topic here on the same subject :
> https://community.platformio.org/t/build-compiles-all-source-files-every-time/11262
> 
> And I found this anomaly:
> https://gavv.github.io/articles/sconsign-bug/
> 
> I tried to modify the line 144 of the file .platformio/penv/lib/python3.7/site-packages/platformio/builder/main.py
> And I replaced:
> `env.SConsignFile( join("$BUILD_DIR", ".sconsign.py%d%d" % (sys.version_info[0], sys.version_info[1])) )`
> by
> `env.SConsignFile( join("$BUILD_DIR", ".sconsign.dblite") )`
> 
> And it looks like if it's working now. Can you try ?

This worked for me, although my file was .platformio\penv\Lib\site-packages\platformio\builder\main.py  at line 149

ivankravets, 2020-03-16:
How did you install PlatformIO Core? it looks like it uses different Pythons in a system.

HeathPetersen, 2020-03-17:
@mastrogippo : If you read the other posts ( see https://github.com/platformio/platform-ststm32/issues/331#issuecomment-569759057 ), you will see that I already tried this.

HeathPetersen, 2020-03-17:
@ivankravets : I may be onto something!  I created a new Linux user and installed using the following command:

`pip install --user platformio`

Then I ran the test case above and **did not** have problems with it recompiling everything.  It worked as expected.

I then created yet another Linux user and and installed using the following command:

`pip3 install --user platformio`

Then I ran the test case above and **did** have problems with it recompiling everything every so often.

I will continue to test using the python2 install and see if I have any further problems.

ivankravets, 2020-03-17:
Please run
```
pip uninstall platformio
pip3 install platformio
```

Use the latest Python 3.

closed 2020-03-17 by ivankravets