platformio/platform-ststm32 #332: [framework-spl] arm-none-eabi-as: unrecognized option '-x'

schwert, 2020-01-01:
Attempting to build any SPL project for F0 fails with the error:
`arm-none-eabi-as: unrecognized option '-x'`
This appears to be caused by `framework-spl/stm32/cmsis/variants/stm32f0/startup_stm32f0xx.s` ending with a lowercase s.
As a result it is sent to arm-none-eabi-as instead of arm-non-eabi-gcc.

Replacing the extension with an uppercase S fixes the problem

positron96, 2020-12-11:
Is it fixed? I've stumbled upon this issue as well. Interestingly, it happens only on linux.

alxpettit, 2022-03-08:
> Is it fixed? I've stumbled upon this issue as well. Interestingly, it happens only on linux.

It only happens on Linux because Linux has a case-sensitive filesystem.