platformio/platform-ststm32 #333: platfrormio_build.py for Arduino STM32Duino core modification required

SuperBoss9, 2020-01-02:
Hello EveryOne,

This is a minor and non critical modification. The problem is in the following:
when a user uses "build.variants_dir" at a board manifest then this script platfrormio_build.py complains that "Warning! Cannot find linker script for the current target!"

That happens because variant_dir variable is created with help of $PROJECT_DIR environment variable. But at linker section variant_dir is checked inside the script where $PROJECT_DIR is not resolved.

So, the solution is in two steps:
1. Adding a new internal variable project_dir = env.subst("$PROJECT_DIR")
2. Modifying variants_dir variable resolving into the following way variants_dir = join(project_dir, board.get("build.variants_dir")) if board.get("build.variants_dir", "") else join(FRAMEWORK_DIR, "variants")

The final code will be:
project_dir = env.subst("$PROJECT_DIR")
variants_dir = join(project_dir, board.get("build.variants_dir")) if board.get("build.variants_dir", "") else join(FRAMEWORK_DIR, "variants")

geeksville, 2020-04-24:
@SuperBoss9  hmm - did this ever get deployed?  I'd very much like to be able to put variants inside my project (rather than in the framework).

SuperBoss9, 2020-04-24:
> @SuperBoss9 hmm - did this ever get deployed? I'd very much like to be able to put variants inside my project (rather than in the framework).

As far as I see it is not fixed yet.