platformio/platform-ststm32 #335: bluepill debugging Please specify `debug_port` for environment

ParamDeshpande, 2020-01-04:
I am using the blue pill and st link v2.
I have successfully uploaded the led blinking code and the printf code.

However, I cant seem to debug the bluepill.
I am getting this error when I try to debug the board.
```
> Executing task: platformio debug <

Error: Please specify `debug_port` for environment
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```
This is what is shown in my lsusb
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 5986:1127 Acer, Inc 
Bus 001 Device 003: ID 0cf3:e500 Atheros Communications, Inc. 
Bus 001 Device 014: ID 0483:374b STMicroelectronics ST-LINK/V2.1 (Nucleo-F103RB)
Bus 001 Device 002: ID 062a:4101 Creative Labs Wireless Keyboard/Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I have connected SWDIO AND SWDCLK properly, also TX and RX from STLINK to A2 and A3 I am using the MBED framework.

this is my default config 
```
// AUTOMATICALLY GENERATED FILE. PLEASE DO NOT MODIFY IT MANUALLY

// PIO Unified Debugger
//
// Documentation: https://docs.platformio.org/page/plus/debugging.html
// Configuration: https://docs.platformio.org/page/projectconf/section_env_debug.html

{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug",
            "executable": "/mnt/Workspace/drone_parachute/.pio/build/genericSTM32F103C8/firmware.elf",
            "toolchainBinDir": "/home/param/.platformio/packages/toolchain-gccarmnoneeabi/bin",
            "svdPath": "/home/param/.platformio/platforms/ststm32/misc/svd/STM32F103xx.svd",
            "preLaunchTask": {
                "type": "PlatformIO",
                "task": "Pre-Debug"
            },
            "internalConsoleOptions": "openOnSessionStart"
        },
        {
            "type": "platformio-debug",
            "request": "launch",
            "name": "PIO Debug (skip Pre-Debug)",
            "executable": "/mnt/Workspace/drone_parachute/.pio/build/genericSTM32F103C8/firmware.elf",
            "toolchainBinDir": "/home/param/.platformio/packages/toolchain-gccarmnoneeabi/bin",
            "svdPath": "/home/param/.platformio/platforms/ststm32/misc/svd/STM32F103xx.svd",
            "internalConsoleOptions": "openOnSessionStart"
        }
    ]
}
```
 and my platformio.int file 
```
;PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = mbed

```

ivankravets, 2020-01-04:
@valeros, please help

SergeySn, 2020-01-16:
I have the same problem with BIGTREETECH-SKR-mini-E3 board.
> Error: Please specify debug_port for environment

How do I specify the debug_port ?

valeros, 2020-01-20:
Hi @ParamDeshpande ! I don't see any mention of `stlink` in your `platformio.ini`. There is no default debug probe specified for `generic*` boards so you need to specify it [manually](http://docs.platformio.org/en/latest/projectconf/section_env_debug.html#debug-tool):
```
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = mbed
debug_tool = stlink
```
Same should work for @SergeySn as well.

SergeySn, 2020-01-20:
@valeros Thanks a lot. Adding the line helped.

> [env:STM32F103RC_bigtree_NOUSB]
> platform          = ststm32
> framework         = arduino
> board             = genericSTM32F103RC
> platform_packages = tool-stm32duino
> extra_scripts     = buildroot/share/PlatformIO/scripts/STM32F103RC_SKR_MINI.py
> build_flags       = !python Marlin/src/HAL/HAL_STM32F1/build_flags.py
>   ${common.build_flags} -DDEBUG_LEVEL=0 -std=gnu++14 -DHAVE_SW_SERIAL -DSS_TIMER=4
> build_unflags     = -std=gnu++11
> lib_deps          = ${common.lib_deps}
>   SoftwareSerialM=https://github.com/FYSETC/SoftwareSerialM/archive/master.zip
> lib_ignore        = SPI
> src_filter        = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
> monitor_speed     = 115200
> **debug_tool        = stlink**
> 

Though now I see a different error:

> Linking .pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf
> /home/serg/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf section `.rodata' will not fit in region `rom'
> /home/serg/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 12252 bytes
> collect2: error: ld returned 1 exit status
> *** [.pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf] Error 1

But that's probably a separate issue.

valeros, 2020-01-24:
Please reopen if you still have any problems with stlink.

closed 2020-01-24 by valeros

jdekarske, 2021-01-29:
This worked for me too, thanks! Any chance this behavior could be default or a warning to do so?