platformio/platform-ststm32 #339: stm32 default core library

nopnop2002, 2020-01-14:
By default, official STM32Duino core is used.
But in the following board maple core is used by default.

	$ cd ~/.platformio/platforms/ststm32/boards
	$ grep -r maple *
	genericSTM32F103C8.json:    "core": "maple",
	genericSTM32F103CB.json:    "core": "maple",
	genericSTM32F103R8.json:    "core": "maple",
	genericSTM32F103RB.json:    "core": "maple",
	genericSTM32F103RC.json:    "core": "maple",
	genericSTM32F103RE.json:    "core": "maple",
	genericSTM32F103T8.json:    "core": "maple",
	genericSTM32F103TB.json:    "core": "maple",
	genericSTM32F103VB.json:    "core": "maple",
	genericSTM32F103VC.json:    "core": "maple",
	genericSTM32F103VD.json:    "core": "maple",
	genericSTM32F103VE.json:    "core": "maple",
	genericSTM32F103ZC.json:    "core": "maple",
	genericSTM32F103ZD.json:    "core": "maple",
	genericSTM32F103ZE.json:    "core": "maple",
	genericSTM32F407VET6.json:    "core": "maple",
	genericSTM32F407VGT6.json:    "core": "maple",
	maple.json:    "core": "maple",
	maple.json:  "url": "http://www.leaflabs.com/maple/",
	maple_mini_b20.json:    "core": "maple",
	maple_mini_b20.json:  "url": "http://www.leaflabs.com/maple/",
	maple_mini_origin.json:  "url": "http://www.leaflabs.com/maple/",
	maple_ret6.json:    "core": "maple",
	maple_ret6.json:  "url": "http://www.leaflabs.com/maple/",
	microduino32_flash.json:    "core": "maple",
	netduino2plus.json:    "core": "maple",
	olimex_p405.json:    "core": "maple",
	stm32f4stamp.json:    "core": "maple",

My Environment:

	$ pio update
	Updating tool-scons                      @ 2.20501.191222 [Up-to-date]

	Platform Manager
	================
	Platform Espressif 32
	--------
	Updating espressif32                     @ 1.11.1         [Up-to-date]
	Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
	Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
	Updating framework-arduinoespressif32    @ 2.10004.191002 [Up-to-date]
	Updating framework-espidf                @ 3.30300.190916 [Up-to-date]
	Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]

	Platform Espressif 32 (Stage)
	--------
	Updating espressif32                     @ cff56ff        [Up-to-date]
	Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
	Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
	Updating framework-espidf                @ 3.30300.190916 [Up-to-date]
	Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]

	Platform Espressif 8266
	--------
	Updating espressif8266                   @ 2.3.2          [Up-to-date]
	Updating tool-esptoolpy                  @ 1.20800.0      [Up-to-date]
	Updating tool-mkspiffs                   @ 1.200.0        [Up-to-date]
	Updating tool-esptool                    @ 1.413.0        [Up-to-date]
	Updating framework-arduinoespressif8266  @ 2.20603.191216 [Up-to-date]
	Updating toolchain-xtensa                @ 2.40802.191122 [Up-to-date]

	Platform ST STM32
	--------
	Updating ststm32                         @ 6.0.0          [Up-to-date]
	Updating tool-stm32duino                 @ 1.0.1          [Up-to-date]
	Updating tool-dfuutil                    @ 1.9.190708     [Up-to-date]
	Updating tool-openocd                    @ 2.1000.190707  [Up-to-date]
	Updating toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]
	Updating framework-arduinoststm32        @ 3.10700.191028 [Up-to-date]

	Library Manager
	===============
	Library Storage: /home/nop/.platformio/lib
	Updating STM32Ethernet                   @ 1.0.5          [Up-to-date]
	Updating STM32duino FreeRTOS             @ 10.0.1         [Up-to-date]
	Updating STM32duino LwIP                 @ 2.1.2          [Up-to-date]

valeros, 2020-02-07:
Hi @nopnop2002 ! That's correct, some boards use `maple` core by default. Is that a problem?

nopnop2002, 2020-02-07:
If you build the following code with Maple Core, you will get a compilation error.
This is a problem of the toolchain used in maple core.

```
#include <Arduino.h>
#define BLINK_DELAY   1000

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  static int state = 0;
  static uint32_t lastupdate = millis();

  uint32_t now = millis();
  if (now - lastupdate >= BLINK_DELAY) {
    state = 1 - state;
    digitalWrite(LED_BUILTIN, state);
    lastupdate = now;
  }
}
```

pfeerick, 2020-05-02:
I think the underlying issue here is partly in relation to the documentation stating that the STM32duino core is [used by default](https://docs.platformio.org/en/latest/platforms/ststm32.html#switching-between-arduino-cores), when it is clearly not. It's understandable for boards that were only defined/supported by the Maple core would use that core by default, but for overlapping boards, and STM32duino only boards, either STM32duino should be the default, or that wording corrected.

nopnop2002, 2020-05-02:
This document has the following description.

https://docs.platformio.org/en/latest/platforms/ststm32.html

```
There are two different Arduino cores for STM32 microcontrollers: STM32Duino and Arduino STM32 (maple). Both of them have been developed independently, therefore, have different functionality and set of internal libraries. By default, official STM32Duino core is used. Some of the boards support both cores. To change the core you can use a board_build.core option that needs be added to build_flags:
```

But that's not true.

utkarshsethi, 2021-05-04:
@nopnop2002 If the issue is with the fact that it is worded wrong, I think someone has already updated the documentation

> Switching between Arduino cores
> ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> 
> There are three different Arduino cores for STM32 microcontrollers: STM32Duino,
> Arduino STM32 (maple) and STM32L0. All of them have been developed independently,
> therefore, have different functionality and set of internal libraries. By default,
> official STM32Duino core is used (except cases when a board supports only one specific
> core). Some of the boards support all three cores. To change the core you can use a
> ``board_build.core`` option that needs be added to :ref:`projectconf_build_flags`:

[https://docs.platformio.org/en/latest/platforms/ststm32.html#switching-between-arduino-cores](url)

nopnop2002, 2021-05-04:
I have confirmed that the official STM32Duino core is used by default.

```
$ cd ~/.platformio/platforms/ststm32/boards
$ grep -r core *
1bitsy_stm32f415rgt.json:    "core": "stm32",
adafruit_feather_f405.json:    "core": "stm32",
afroflight_f103cb.json:    "core": "stm32",
armed_v1.json:    "core": "stm32",
armstrap_eagle1024.json:    "core": "stm32",
armstrap_eagle2048.json:    "core": "stm32",
armstrap_eagle512.json:    "core": "stm32",
b96b_aerocore2.json:    "core": "stm32",
b96b_aerocore2.json:      "variant": "96b_aerocore2"
b96b_aerocore2.json:  "url": "https://www.96boards.org/product/aerocore2/",
black_f407ve.json:    "core": "stm32",
black_f407vg.json:    "core": "stm32",
black_f407ze.json:    "core": "stm32",
black_f407zg.json:    "core": "stm32",
blackpill_f103c8.json:    "core": "stm32",
blackpill_f103c8_128.json:    "core": "stm32",
blackpill_f401cc.json:    "core": "stm32",
blackpill_f401ce.json:    "core": "stm32",
blackpill_f411ce.json:    "core": "stm32",
blue_f407ve_mini.json:    "core": "stm32",
bluepill_f103c6.json:    "core": "stm32",
bluepill_f103c8.json:    "core": "stm32",
bluepill_f103c8_128k.json:    "core": "stm32",
cicada_l082cz.json:    "core": "stm32l0",
cloud_jam.json:    "core": "stm32",
coreboard_f401rc.json:    "core": "stm32",
cricket_l082cz.json:    "core": "stm32l0",
disco_f303vc.json:    "core": "stm32",
disco_f407vg.json:    "core": "stm32",
disco_f413zh.json:    "core": "stm32",
disco_g031j6.json:    "core": "stm32",
disco_l100rc.json:    "core": "stm32",
disco_l152rb.json:    "core": "stm32",
diymore_f407vgt.json:    "core": "stm32",
econode_l082cz.json:    "core": "stm32l0",
electrosmith_daisy.json:    "core": "stm32",
fk407m1.json:    "core": "stm32",
fysetc_s6.json:    "core": "stm32",
genericSTM32F103C4.json:    "core": "stm32",
genericSTM32F103C6.json:    "core": "stm32",
genericSTM32F103C8.json:    "core": "stm32",
genericSTM32F103CB.json:    "core": "stm32",
genericSTM32F103R4.json:    "core": "stm32",
genericSTM32F103R6.json:    "core": "stm32",
genericSTM32F103R8.json:    "core": "stm32",
genericSTM32F103RB.json:    "core": "stm32",
genericSTM32F103RC.json:    "core": "stm32",
genericSTM32F103RD.json:    "core": "stm32",
genericSTM32F103RE.json:    "core": "stm32",
genericSTM32F103RF.json:    "core": "stm32",
genericSTM32F103RG.json:    "core": "stm32",
genericSTM32F103T4.json:    "core": "stm32",
genericSTM32F103T6.json:    "core": "stm32",
genericSTM32F103T8.json:    "core": "stm32",
genericSTM32F103TB.json:    "core": "stm32",
genericSTM32F103V8.json:    "core": "stm32",
genericSTM32F103VB.json:    "core": "stm32",
genericSTM32F103VC.json:    "core": "stm32",
genericSTM32F103VD.json:    "core": "stm32",
genericSTM32F103VE.json:    "core": "stm32",
genericSTM32F103VF.json:    "core": "stm32",
genericSTM32F103VG.json:    "core": "stm32",
genericSTM32F103ZC.json:    "core": "stm32",
genericSTM32F103ZD.json:    "core": "stm32",
genericSTM32F103ZE.json:    "core": "stm32",
[... 64 lines cut ...]
nucleo_h745zi_q.json:    "core": "stm32",
nucleo_l433rc_p.json:    "core": "stm32",
olimex_e407.json:    "core": "stm32",
olimex_f103.json:    "core": "stm32",
olimex_h407.json:    "core": "stm32",
olimex_p405.json:    "core": "stm32",
olimexino.json:    "core": "stm32",
portenta_h7_m4.json:    "core": "arduino",
portenta_h7_m4.json:  "name": "Arduino Portenta H7 (M4 core)",
portenta_h7_m7.json:    "core": "arduino",
portenta_h7_m7.json:  "name": "Arduino Portenta H7 (M7 core)",
prntr_v2.json:    "core": "stm32",
pybstick26_duino.json:    "core": "stm32",
pybstick26_lite.json:    "core": "stm32",
pybstick26_pro.json:    "core": "stm32",
pybstick26_std.json:    "core": "stm32",
rhf76_052.json:    "core": "stm32",
robotdyn_blackpill_f303cc.json:    "core": "stm32",
sparky_v1.json:    "core": "stm32",
st3dp001_eval.json:    "core": "stm32",
stm32f4stamp.json:    "core": "stm32",
thunder_pack_f411.json:    "core": "stm32",
vccgnd_f103zet6.json:    "core": "stm32",
waveshare_open103z.json:    "core": "stm32",
```

closed 2021-05-04 by nopnop2002