platformio/platform-ststm32 #340: STM32F303CC: Linker script doesn't know about 8KByte CCM RAM

maxgerhardt, 2020-01-17:
As one can read in the discussion at https://community.platformio.org/t/robotdyn-blackpill-stm32f303cc-pio-has-wrong-max-ram-size/11521, the STM32F303CC chip (for which we have the board https://github.com/platformio/platform-ststm32/blob/develop/boards/robotdyn_blackpill_f303cc.json) has wrong RAM size information. Specifically, the board has 40KB SRAM (starting at 0x20000000) and 8KB of core-coupled-memory (CCM) starting at 0x10000000. The board definition just says 40KB and the used linker script (https://github.com/platformio/platform-ststm32/blob/master/ldscripts/stm32f30xx.ld#L152) references some completely "mysterios memory bank 1", which is only readable and executable, and of 0 length. So no mention of CCM_RAM at 0x10000000. 

The linker script should be fixed to allow the creation of a section in which both code and data can be placed, at the correct address. User code will then have to use `__attribute__ ((section ("ccm_ram")))`-like instructions to place stuff in this linker memory section.

For now, the CCM RAM cannot be used with PIO.

valeros, 2020-01-20:
Thanks for the report. Extra memory sections is a known issue and there is no default solution for this case at the moment, besides as I can see from the discussion that Arduino framework is used which means we don't have control over the default linker scripts.

A possible workaround is to manually increase RAM size in `platformio.ini` file using `board_upload.maximum_ram_size = 49152`.

valeros, 2020-01-24:
If I'm not mistaken, `stm32f30xx.ld` is a legacy linker script and it not used by any of the frameworks anymore.

Alteregoxxx, 2020-02-04:
> Thanks for the report. Extra memory sections is a known issue and there is no default solution for this case at the moment, besides as I can see from the discussion that Arduino framework is used [...]

Oh, this would be actually great! Does this workaround still need that I have to explicitly use __attribute__ GCC to use to that part of "special SRAM" (CCRAM) or is not necessary because it will be automatically managed behind the scenes by PlatformIO ?

valeros, 2020-02-04:
@Alteregoxxx I'm not sure I understood your question correctly, but PlatformIO relies on `maximum_ram_size` value when checks whether the firmware will fit in your device, that's it. That workaround only bypasses this limitation. If you want to use this extra RAM section you need to specify `attribute` keyword and if I'm not mistaken you also need to modify startup routine because initial values for variables in CCRAM need to be copied from Flash when your device boots up.

Alteregoxxx, 2020-02-04:
> @Alteregoxxx I'm not sure I understood your question correctly, but PlatformIO relies on `maximum_ram_size` value when checks whether the firmware will fit in your device, that's it. That workaround [...]

In the meantime I did a little bit of experimentation and it seems that only using the `attribute` keyword is sufficient to allow successful build of a firmware bigger than 40KB; hence, at least apparently, it seems it's not necessary to use `board_upload.maximum_ram_size = 49152` in platformio.ini
Later I will try also if the firmware runs correctly on the board 

Thank you!