platformio/platform-ststm32 #341: Error: `.rodata' will not fit in region `rom'

SergeySn, 2020-01-20:
I am trying to debug Marlin code for BigTreeTech SKR Mini E3 board.  When I click the build button - it compiles fine.  And when I click the "Platform IO: Upload" button it compiles and uploads via STLINK fine. But when I click "Start Debugging" menu, it gives me an error:

> Executing task: platformio debug 
> Processing STM32F103RC_bigtree_NOUSB (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
> PLATFORM: ST STM32 6.0.0 > STM32F103RC (48k RAM. 256k Flash)
> HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
> DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
> PACKAGES: framework-arduinoststm32-maple 1.10000.190819 (1.0.0), toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), tool-stm32duino 1.0.1
> Converting Marlin.ino
> LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 39 compatible libraries
> Scanning dependencies...
> Dependency Graph
> ...
> Building in debug mode
> Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/board.cpp.o
> Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/wirish/boards.cpp.o
> Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/wirish/boards_setup.cpp.o
> ...
> Compiling .pio/build/STM32F103RC_bigtree_NOUSB/lib479/EEPROM/EEPROM.cpp.o
> Compiling .pio/build/STM32F103RC_bigtree_NOUSB/lib479/EEPROM/flash_stm32.c.o
> Archiving .pio/build/STM32F103RC_bigtree_NOUSB/lib479/libEEPROM.a
> Indexing .pio/build/STM32F103RC_bigtree_NOUSB/lib479/libEEPROM.a
> Linking .pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf
> /home/serg/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf **section `.rodata' will not fit in region `rom'**
> /home/serg/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 12244 bytes
> collect2: error: ld returned 1 exit status
> [.pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf] Error 1

What might be wrong?

valeros, 2020-01-24:
As it can be seen from the log, your project requires too much flash memory when built in debug mode. Recently, we've added the ability to modify the default build flags for debug mode. If you want to use it you need to switch to the latest platformio-core. Just run the following command in the terminal window: `pio upgrade --dev`. Now you should be able to add the following line to your environment:
```
platform = ...
board = ...
debug_build_flags = -Os -ggdb2 -g2
```

Please keep in mind that this can worsen your debug experience since a lot of optimizations will be enabled, but at least you'll be able to inspect your project.

SergeySn, 2020-01-24:
@valeros Thanks a lot for the great info.  Now I understand that when I click "Start Debugging" menu - it automagically makes the code size is larger, without me changing the "-DDEBUG_LEVEL=0".  
So, I removed some of the code and now it fits  :-) 
Again, thanks so much.

closed 2020-01-24 by SergeySn