platformio/platform-ststm32 #342: Error when uploading to nucleo_wb55rg_p

vshymanskyy, 2020-01-20:
```log
CURRENT: upload_protocol = stlink
openocd -d2 -s /home/vshymanskyy/.platformio/packages/tool-openocd/scripts -f interface/stlink.cfg -c "transport select hla_swd" -f target/stm32w55xx.cfg -c "program {.pio/build/Nucleo_WB55RG/firmware.elf}  verify reset; shutdown;"
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

hla_swd
embedded:startup.tcl:26: Error: Can't find target/stm32w55xx.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
*** [upload] Error 1
```

valeros, 2020-01-21:
Hi @vshymanskyy, thanks for the report.  Do you see a mass storage device when the board is connected to you your computer?

vshymanskyy, 2020-01-21:
Yes, but pio isn't even trying to upload - the necessary files are missing.

valeros, 2020-01-22:
Could you please try uploading via this mass storage device? Just add this line to your project environment:
```
[env:nucleo_wb55rg_p]
...
upload_protocol = mbed
```
Thanks!

vshymanskyy, 2020-01-22:
@valeros it worked! ;) THANKS

closed 2020-01-22 by vshymanskyy