platformio/platform-ststm32 #343: Debugger is not working with STLINK

SergeySn, 2020-01-25:
I am trying to debug Marlin code for BIGTREETECH-SKR-mini-E3 board. 
But when I click the debug button - it goes to Start.S file line 51:
> ldr r1,=__msp_init

shows the current execution point there for a second, and then exits the debugger.
In debug console it shows the following:

```
Reading symbols from /a/proj/VSCode/BIGTREETECH-SKR-mini-E3-/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/.pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf...
done.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = stlink
PlatformIO: Initializing remote target...
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:25)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : tcl server disabled
Info : telnet server disabled
Info : clock speed 1000 kHz
Info : STLINK V2J32M22 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.254104
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection from pipe
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x00000003 pc: 00000000 msp: 0x464c454c
Info : device id = 0x10036414
Info : flash size = 256kbytes
0x00000000 in ?? ()
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Loading section .text, size 0x18a24 lma 0x8007000
Loading section .ARM.exidx, size 0x8 lma 0x801fa28
Loading section .data, size 0x1058 lma 0x801fa30
Loading section .rodata, size 0x176c lma 0x8020a88
Info : Padding image section 0 at 0x0801fa24 with 4 bytes
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Start address 0x80073f8, load size 111088
Transfer rate: 18 KB/sec, 11108 bytes/write.
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Temporary breakpoint 1 at 0x800ab16: file /home/serg/.platformio/packages/framework-arduinoststm32-maple/STM32F1/cores/maple/main.cpp, line 38.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> http://bit.ly/pio-debug
Note: automatically using hardware breakpoints for read-only addresses.
```
What might be wrong? 
Any help is greatly appreciated.

valeros, 2020-01-25:
It's hard to guess what exactly is wrong here. I don't see anything suspicious in the log. Are you sure you aren't using JTAG/SWD pins in your project? Could you please try disabling the initial breakpoint? Just add the following line to your project environment:
```
platform = ...
board = ...

; Disable initial breakpoint
debug_init_break =
```
Debug session should stop at `dr r1,=__msp_init` and the you'd be able to manually proceed to the `main` function using `Step over` commands.

SergeySn, 2020-01-26:
@valeros Thanks a lot for the answer. 
I don't know if the code uses JTAG/SWD pins.  I didn't write the code. I have downloaded it from https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3
But I don't think it should matter, because the code have not started executing yet.  It didn't even get to the main() function.
 I have added the `debug_init_break =` line and it did stop at the 
`dr r1,=__msp_init`
line, as you have predicted. Then, I have started to step over. I pressed F10 about 3 times, and saw this in the log:
```
Info : Target voltage: 3.249376
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection from pipe
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x00000003 pc: 00000000 msp: 0x464c454c
Info : device id = 0x10036414
Info : flash size = 256kbytes
0x00000000 in ?? ()
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Loading section .text, size 0x18a24 lma 0x8007000
Loading section .ARM.exidx, size 0x8 lma 0x801fa28
Loading section .data, size 0x1058 lma 0x801fa30
Loading section .rodata, size 0x176c lma 0x8020a88
Info : Padding image section 0 at 0x0801fa24 with 4 bytes
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
Start address 0x80073f8, load size 111088
Transfer rate: 18 KB/sec, 11108 bytes/write.
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00010100 msp: 0x464c457c
PlatformIO: Initialization completed
Note: automatically using hardware breakpoints for read-only addresses.
Info : halted: PC: 0x00010044
halted: PC: 0x00010044
Info : halted: PC: 0x00000000
halted: PC: 0x00000000
Info : halted: PC: 0x00000000
halted: PC: 0x00000000
Info : halted: PC: 0x00000000
halted: PC: 0x00000000
```
It looks like it's stuck at 0x00000000 and not moving any further. With each additional step, it just prints another pair of lines:
```
Info : halted: PC: 0x00000000
halted: PC: 0x00000000
```
There is another issue - this code used to run good when I was flushing it via the sd-card. 
But when I upload it via STLINK - it doesn't work.  (That's why I am trying to debug it.)
But now I wonder if the debugger is not working for the same reason?

valeros, 2020-01-27:
@SergeySn  The only possible reason I came up with is that there might be a bootloader on this board, so you need to specify an offset address for your application. I'm not sure about the value for this offset, but you can look it up in the linker file used for the board.
```
platform = ...
board = ...
board_upload.offset_address = 0x08007000
```

SergeySn, 2020-01-27:
How do I specify the offset_address?
What's "linker file", is it "platformio.ini" ?
It doesn't have "board_upload.offset_address".  Also I can't find "offset_address" anywhere in the project.
But I do see:
```
# Relocate firmware from 0x08000000 to 0x08007000
env['CPPDEFINES'].append(("VECT_TAB_ADDR", "0x08007000"))
```
in STM32F103RC_SKR_MINI.py
Is it correct?

SergeySn, 2020-01-27:
@valeros I have added your line:
`board_upload.offset_address = 0x08007000`
to platformio.ini and now, in release mode, it uploads my firmware to 0x08007000 address, and doesn't override the bootloader anymore.  
Also it looks like this line has fixed something else: 
Without this line, the code at 0x08000000 started with 
`7f 45 4c 46   ELF`
and with this line, the code at 0x08007000 starts with
`2000BFD8    ???`
I wonder why?

So, in release mode, without this line, the code was uploaded to the beginning of the memory, and it had a wrong start "?ELF", that's why it didn't work.
In debug mode, regardless of this line, the code uploads to 0x08007000, but without the bootloader, (which was overwritten by "?ELF...") it didn't work either.
So, with this line, and the bootloader, both release and debug works fine.
So, this issue is solved.  Thanks a lot.

P.S.  I still wonder where this "?ELF" came from, and why it disappeared when this line was added?

closed 2020-01-31 by SergeySn