platformio/platform-ststm32 #344: More CPU variants support with libopencm3

JackDesBwa, 2020-01-25:
libopencm3 support is currently (very) outdated in PlatformIO (last update in 2016) and the effort to get access to the full CPU variants it supports now is pretty low. It works on my station so it would be nice to integrate it in the framework.

I submitted an issue comment in builder-framework-libopencm3 repo and a message on community forum a few months ago without any answer. Ivan Kravets just advised to post here too.

I have not as much time as I had when I submitted them, so I will probably not be able to do most of the work myself (except if the blocking points are explained and I have time in the future), but I can answer questions if any.
Anyways, all the code and the little improvements I identified (but have not enough skills to do completely, mostly removal of hardcoded executable paths) to integrate it correctly in the framework are documented. See https://community.platformio.org/t/update-libopencm3-wip-help-needed/10544 for details.

tute-avalos, 2020-06-25:
I was comment at an older issue: https://github.com/platformio/platform-ststm32/issues/218
The tar.gz simes to be in the PlatformIO Storage project at SF: https://sourceforge.net/projects/platformio-storage/files/packages/
Don't know how to contribute there to upload the file and update the manifest.json, someone?

valeros, 2020-08-03:
Resolved in https://github.com/platformio/platform-ststm32/commit/0bf8d98f1944d76e4b8159962c04969d610c30ab. Please update your platform via `pio platform update`.

closed 2020-08-03 by valeros

JackDesBwa, 2020-08-05:
I checked on an old project, it works.