platformio/platform-ststm32 #345: New release with latest stm32 arduino core

ramboerik, 2020-01-26:
Hi, 

The [stm32 arduino core](https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/1.8.0)  was updated a while ago and I wonder when the next release including those changes is  planned? 

Thanks for great work with platformio!

ramboerik, 2020-01-29:
A more general question about pio + stm's arduino core:
Why isn't the pio build files(`packages.json` and  `tools/platformio-build.py`) merged into stm's arduino repo? I would like to configure `platformio.ini` to point directly to the repo but it's currently not supported as the pio files are kept out of the repo.

My non-working configuration:
```
[env:nucleo_f429zi]
platform = ststm32
board = nucleo_f429zi
framework = arduino
platform_packages = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32
```

fpistm, 2020-01-30:
@ivankravets 
I saw several time that PIO needs to update manually each release.
I don't know exactly how it is handle anyway, if adding some config/script files to the core helps on this subject do not hesitate to provide a PR.

ivankravets, 2020-01-30:
@fpistm thanks for the proposition with PR. We didn't think that you can merge it. @valeros will provide the integration for the latest release and will prepare a PR to your repository. Thanks!

P.S: We did the same for Espressif repositories. Now, develoeprs can use even dev-branch for ESP32/ESP8266.

ramboerik, 2020-01-30:
Perfect! I think this would be a great improvement for both pio and stm 😃

fpistm, 2020-01-30:
> We didn't think that you can merge it.

Welcome, I can merge what I want :wink:

valeros, 2020-02-03:
Hi @fpistm ! I can submit a PR, where is the right place for our build script in your repository?

fpistm, 2020-02-03:
Hi  @valeros 
Well currently, I've only a CI folder for all related things about Continuous Integration.
So, I wonder if your build script can be considered like that and added to this one in a certain way this allow CI to PIO.
In other cores, where it was added?

valeros, 2020-02-03:
@fpistm I'm not sure the `CI` folder is the right place. Usually our build script `platformio-build.py` is located in the folder `tools` (for example in [esp8266](https://github.com/esp8266/Arduino/tree/master/tools) and [esp32](https://github.com/espressif/arduino-esp32/tree/master/tools) cores).

fpistm, 2020-02-03:
Ok so, I guess you could create a `tools/` folder.

ramboerik, 2020-03-27:
@valeros 

To start with, thanks for your work on this.

When I use pio and point directly to the [st' arduino core](https://github.com/stm32duino/Arduino_Core_STM32) I cannot build and get the following assert error from pio:
```
File "/home/erik/.platformio/packages/framework-arduinoststm32@src-7fbfdaa89296b31ecfb0fcc25f6a0b73/tools/platformio-build.py", line 37:
    assert isdir(CMSIS_DIR)
    assert isdir(CMSIS_DIR)
```

The CMSIS directory is missing as it's not a part of the repo(should maybe be a submodule?) and no other actions is taken to download it. A  temporary workaround is to manually copy the CMSIS directory into `~/.platformio/packages/framework-arduinoststm32/CMSIS`

My platformio.ini:
```ini
[env:nucleo_f429zi]
platform = ststm32
board = nucleo_f429zi
framework = arduino
platform_packages = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32
```