platformio/platform-ststm32 #346: Compilation error with framework = cmsis and upload_protocol = dfu

Trance-Paradox, 2020-02-01:
platformio.ini:
[env:genericSTM32F103C8]
platform = ststm32
framework = cmsis
board = genericSTM32F103C8
upload_protocol = dfu

error:
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [          ]   0.0% (used 0 bytes from 20480 bytes)
PROGRAM: [          ]   0.4% (used 232 bytes from 65536 bytes)
Adding dfu suffix to firmware.bin
'C:\Users\Surya' is not recognized as an internal or external command,
operable program or batch file.
*** [.pio\build\genericSTM32F103C8\firmware.bin] Error 1
============================================== [FAILED] Took 1.55 seconds ==============================================

valeros, 2020-02-03:
Hi @Trance-Paradox ! Looks like you have a whitespace character in your username?

Trance-Paradox, 2020-02-03:
Umm its a small dash

valeros, 2020-02-03:
Could you please compile your project in verbose mode and attach the log here?

Trance-Paradox, 2020-02-04:
I can run it in verbose mode but how to store it in a log file?

Trance-Paradox, 2020-02-04:
[build.log](https://github.com/platformio/platform-ststm32/files/4154691/build.log)
is this what you wanted

Trance-Paradox, 2020-02-06:
Has anyone checked on it yet?

Trance-Paradox, 2020-02-06:
I get it now cli is not recognizing my user name folder with space in it 
How can i fix that issue.

closed 2020-02-07 by valeros

Trance-Paradox, 2020-02-07:
I tried upload_command in the ini folder and put the directory in ' '. It worked but it would be nise it it was included in main file

commit 03241176 (2020-02-07): Handle possible spaces in path to dfu tool // Resolve #346