platformio/platform-ststm32 #349: dfu upload id mismatch while uploading

Trance-Paradox, 2020-02-08:
"No DFU capable USB device available"
This is the error while using cmsis

platformio.ini:

[env:genericSTM32F103C8]
platform = ststm32
framework = cmsis
board = genericSTM32F103C8
upload_protocol = dfu

However this problem doesn't arise when framework = arduino is used. It uploads and resets fine then.
[upload.log](https://github.com/platformio/platform-ststm32/files/4174588/upload.log)

laveur, 2020-02-26:
I have a similar issue with my config:

```
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = mbed
upload_protocol = dfu
```

dharders, 2020-03-04:
Is there any current workaround?

I'm hitting this too with similar config. (Arduino and STM32duino bootloader working OK though with blinky). PlatformIO, cmsis combination builds OK but fails upload looking for 1eaf:0003, but only 1eaf:0004 available. Looks like platformio not soft triggering bluepill, thus remaining in USB serial mode.

Would love a workaround if anyone knows of one?