platformio/platform-ststm32 #351: Serial support for blue and black pill

Trance-Paradox, 2020-02-20:
Can you please add support for bluepill and blackpill stm32f103c8 
Upload protocol = serial

lnlp, 2020-03-05:
I would like that too, but it will require a custom bootloader like https://github.com/rogerclarkmelbourne/STM32duino-bootloader.
`upload_protocol = serial` will have to be implemented specifically for that bootloader.

Trance-Paradox, 2020-03-06:
I mean serial not serial over usb

> Do you mean serial or serial over USB? IIRC latter would require `upload_protocol = dfu`.
> stm32f103c8 does not support this by default and will require a custom bootloader like https://github.com/rogerclarkmelbourne/STM32duino-bootloader.
> The upload protocol will then be dependant on the custom bootloader.

jnr2820, 2020-04-25:
Hi @Trance-Paradox ,
If you are referring to uploading code to a Blackpill such as the one from RobotDyn, with the "arduino" boot loader that it ships with, that function is working in current build. 
I was able to do a code upload by using an FTDI serial to USB adapter, connected directly to the serial pins (9 and 10 on mine) of the board. Remember to flip the jumper closest to the USB port to "1" and reset to change the mode to allow uploads, first.

Here's the relevant piece from my platformio.ini. Note that both "upload_port" and "upload_protocol" seem to be required. The "upload_port" below is the address of my FTDI adapter as detected by my Macbook.

`[env:blackpill]
platform = ststm32
framework = arduino
board = blackpill_f103c8_128
upload_port = /dev/cu.usbserial-AD0JVIWC
upload_protocol = serial`