platformio/platform-ststm32 #354: Unknown board ID 'olimex_e407'", but the board was mentioned in docs

N0rbert, 2020-03-06:
Steps to reproduce:
1. Have [Olimex STM32-E407 board](https://www.olimex.com/Products/ARM/ST/STM32-E407)
1. Read the docs on https://docs.platformio.org/en/latest/boards/ststm32/olimex_e407.html
1. Try to create new project with this board

Expected results:
* project was created

Actual results:
* project is not created, got the following messages:

   >PIO Core Call Error: "The current working directory /home/user/Documents/PlatformIO/Projects/ee407 will be used for the project.\n\nThe next files/directories have been created in /home/user/Documents/PlatformIO/Projects/ee407\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Unknown board ID 'olimex_e407'"

valeros, 2020-04-02:
Hi @N0rbert ! What version of the platform is installed in your system?

N0rbert, 2020-04-02:
@valeros  I have PlatformIO - Core 4.2.1,·Home 3.1.1

valeros, 2020-04-02:
I meant the version of `ststm32` platform:
`platformio platform show ststm32`

N0rbert, 2020-04-02:
ststm32 is 6.0.0.

valeros, 2020-04-03:
Strange, but I cannot reproduce the issue. Could you please try to create a project for any other board and then manually specify `olimex_e407` in `platformio.ini` file:
```ini
[env:olimex_e407]
platform = ststm32
board = olimex_e407
framework = arduino
```

N0rbert, 2020-04-03:
Did clean reinstallation of PlatformIO with VSCodium and problem solved.

closed 2020-04-03 by N0rbert