platformio/platform-ststm32 #355: GCC 5.4.1: `arm-none-eabi-ar` missing

stertingen, 2020-03-07:
My `platformio.ini:`

```ini
[env:disco_f051r8]
platform = ststm32
board = disco_f051r8
framework = stm32cube
platform_packages =
    toolchain-gccarmnoneeabi@~1.50401.0
```

Build log:
```
Processing disco_f051r8 (platform: ststm32; board: disco_f051r8; framework: stm32cube)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_f051r8.html
PLATFORM: ST STM32 6.0.0 > ST STM32F0DISCOVERY
HARDWARE: STM32F051R8T6 48MHz, 8KB RAM, 64KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES: 
 - framework-stm32cube 2.0.181130 
 - toolchain-gccarmnoneeabi 1.50401.190816 (5.4.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/disco_f051r8/src/main.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_adc.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_adc_ex.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_can.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_cec.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_comp.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_cortex.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_crc.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_crc_ex.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_dac.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_dac_ex.o
/home/hermann/.platformio/packages/framework-stm32cube/f0/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_crc.c: In function 'CRC_Handle_8':
/home/hermann/.platformio/packages/framework-stm32cube/f0/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_crc.c:475:8: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
        *(uint16_t volatile*) (&hcrc->Instance->DR) = ((uint32_t)pBuffer[4*i]<<8) | (uint32_t)pBuffer[4*i+1];
        ^
/home/hermann/.platformio/packages/framework-stm32cube/f0/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_crc.c:479:8: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
        *(uint16_t volatile*) (&hcrc->Instance->DR) = ((uint32_t)pBuffer[4*i]<<8) | (uint32_t)pBuffer[4*i+1];
        ^
/home/hermann/.platformio/packages/framework-stm32cube/f0/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_crc.c: In function 'CRC_Handle_16':
/home/hermann/.platformio/packages/framework-stm32cube/f0/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_crc.c:511:8: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
        *(uint16_t volatile*) (&hcrc->Instance->DR) = pBuffer[2*i]; 
        ^
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_dma.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_flash.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_flash_ex.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_gpio.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_i2c.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_i2c_ex.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_i2s.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_irda.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_iwdg.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_pcd.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_pcd_ex.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_pwr.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_pwr_ex.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_rcc.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_rcc_ex.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_rtc.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_rtc_ex.o
Compiling .pio/build/disco_f051r8/FrameworkHALDriver/Src/stm32f0xx_hal_smartcard.o
[... 105 lines cut ...]
-rwxr-xr-x 2 root root  776368 Jun 22  2016 arm-none-eabi-gcc-5.4.1
-rwxr-xr-x 1 root root   24324 Jun 22  2016 arm-none-eabi-gcc-ar
-rwxr-xr-x 1 root root   24292 Jun 22  2016 arm-none-eabi-gcc-nm
-rwxr-xr-x 1 root root   24324 Jun 22  2016 arm-none-eabi-gcc-ranlib
-rwxr-xr-x 1 root root  447252 Jun 22  2016 arm-none-eabi-gcov
-rwxr-xr-x 1 root root  418572 Jun 22  2016 arm-none-eabi-gcov-tool
-rwxr-xr-x 1 root root 4526856 Jun 22  2016 arm-none-eabi-gdb
-rwxr-xr-x 1 root root 5004552 Jun 22  2016 arm-none-eabi-gdb-py
-rwxr-xr-x 1 root root  837764 Jun 22  2016 arm-none-eabi-gprof
-rwxr-xr-x 2 root root  777136 Jun 22  2016 arm-none-eabi-g++
-rwxr-xr-x 4 root root 1091400 Jun 22  2016 arm-none-eabi-ld
-rwxr-xr-x 4 root root 1091400 Jun 22  2016 arm-none-eabi-ld.bfd
-rwxr-xr-x 2 root root  784196 Jun 22  2016 arm-none-eabi-nm
-rwxr-xr-x 2 root root  953668 Jun 22  2016 arm-none-eabi-objcopy
-rwxr-xr-x 2 root root 1172484 Jun 22  2016 arm-none-eabi-objdump
-rwxr-xr-x 2 root root  800200 Jun 22  2016 arm-none-eabi-ranlib
-rwxr-xr-x 2 root root  464724 Jun 22  2016 arm-none-eabi-readelf
-rwxr-xr-x 1 root root  773188 Jun 22  2016 arm-none-eabi-size
-rwxr-xr-x 1 root root  772772 Jun 22  2016 arm-none-eabi-strings
-rwxr-xr-x 2 root root  953668 Jun 22  2016 arm-none-eabi-strip
```

So I guess the package delivered as `toolchain-gccarmnoneeabi` for version `~1.50401.0` is broken.

Does anyone know where the toolchain package is generated for PlatformIO?

valeros, 2020-03-10:
Hi @stertingen ! What is your OS ?

stertingen, 2020-03-10:
> Hi @stertingen ! What is your OS ?

Up-to-date Arch linux; PlatformIO installed via AUR. (https://aur.archlinux.org/packages/platformio)

valeros, 2020-03-10:
And the arhitecture? x64? ARM?

stertingen, 2020-03-10:
> And the arhitecture? x64? ARM? BTW, is there a reason why you need 5.4.1?

It's x86_64.

GCC 5.4.1 is convention in a student robotics group due to compiler bugs in later versions. While evaluating whether PlatformIO might be a good choice for embedded development, I stumbled upon this bug.

And since PlatformIO ships GCC 5.4.1, this package should not produce any errors.

Can you tell me who maintains the GCC package for PlatformIO? I'd also like to file a bug report there.

valeros, 2020-03-10:
This package is maintained internally, so there is no repository for it. Also, there are no guarantees that this package is compatible with `ststm32` since it's meant to be used only with the `teensy` platform and for some reason `arm-none-eabi-ar` is replaced with `arm-none-eabi-gcc-ar`.

stertingen, 2020-03-13:
> This package is maintained internally, so there is no repository for it.

Alright, so the chances that this will be fixed are low.

> for some reason `arm-none-eabi-ar` is replaced with `arm-none-eabi-gcc-ar`.

No, it's not replaced. `arm-none-eabi-gcc-ar` is always there in the `bin` directory. It's just `arm-none-eabi-ar` that's missing. In version 6.3.1, it's a symlink to `../arm-none-eabi/ar`, in version 5.4.1, the symlink does not exist, but the file it would point to.