platformio/platform-ststm32 #358: Nucleo L412KB upload fails

microbug, 2020-03-17:
[This issue (forum link)](https://community.platformio.org/t/default-linker-script-fails-on-stm32l412-fix-included/6019/6) is still affecting me with the latest stable .

Summary:
Uploading to the L412KB using the built-in ST-Link does not work. Log:

```
> Executing task in folder rack-fan-controller: platformio run --verbose --target upload <

Processing nucleo_l412kb (platform: ststm32; board: nucleo_l412kb; framework: arduino)

CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_l412kb.html
PLATFORM: ST STM32 5.5.0 > ST Nucleo L412KB
HARDWARE: STM32L412KBU6 80MHz, 40KB RAM, 128KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES: 
 - framework-arduinoststm32 3.10601.190716 (1.6.1) 
 - tool-dfuutil 1.9.190708 
 - tool-openocd 2.1000.190422 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/nucleo_l412kb/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   2.0% (used 828 bytes from 40960 bytes)
Flash: [=         ]   8.6% (used 11324 bytes from 131072 bytes)
.pio/build/nucleo_l412kb/firmware.elf  :
section              size        addr
.isr_vector           396   134217728
.text               10476   134218124
.rodata               812   134228600
.init_array            16   134229412
.fini_array             4   134229428
.data                  36   536870912
.bss                  792   536870948
._user_heap_stack    1536   536871740
.ARM.attributes        48           0
.comment              126           0
.debug_frame          148           0
Total               14390
<lambda>(["upload"], [".pio/build/nucleo_l412kb/firmware.elf"])
AVAILABLE: blackmagic, jlink, stlink
CURRENT: upload_protocol = stlink
openocd -d2 -s /Users/richard/.platformio/packages/tool-openocd/scripts -f board/st_nucleo_l4.cfg -c "program {.pio/build/nucleo_l412kb/firmware.elf}  verify reset; shutdown;"
GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-23-00:01)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 500 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : clock speed 500 kHz
Info : STLINK V2J31M21 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.248627
Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
adapter speed: 480 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08004604 msp: 0x200022d0
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
Info : device id = 0x10006464
Warn : Cannot identify target as an STM32L4 family device.
Error: auto_probe failed
embedded:startup.tcl:479: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 538
at file "embedded:startup.tcl", line 479
*** [upload] Error 1
```

Upload works fine using STM32CubeProgrammer with Arduino.

microbug, 2020-03-17:
Building OpenOCD from source solves this for me. I've also tested with the OpenOCD from Brew (I'm on macOS) but that didn't work.

Building OpenOCD from source is fairly simple. Once you have the `openocd` binary, go to `~/.platformio/packages/tool-openocd/bin`, rename `openocd` to `openocd.old` and copy in the freshly built `openocd` binary.

Tested with `Open On-Chip Debugger 0.10.0+dev-01135-gdca1c6ca-dirty (2020-03-17-00:33)`

microbug, 2020-03-17:
I've tested using the development version of PlatformIO and the git version of `platform-ststm32`, with the same result. The only thing that works is building my own OpenOCD.

I hope this could be solved easily by rebuilding the OpenOCD in JFrog Bintray.

valeros, 2020-08-03:
The OpenOCD package has been updated, please update packages via `pio update` and retest.

closed 2020-08-03 by valeros