platformio/platform-ststm32 #363: nucleo f446ze: target halted due to debug-request

ChrisIdema, 2020-03-25:
I have a platformio project in vscode. I use the source code of the blink example: https://github.com/platformio/platform-ststm32/tree/master/examples/stm32cube-hal-blink .
The project builds, but when I try to flash it I get the following error:

> Checking size .pio\build\nucleo_f446ze\firmware.elf
> Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
> RAM:   [          ]   0.2% (used 268 bytes from 131072 bytes)
> Flash: [          ]   0.5% (used 2484 bytes from 524288 bytes)
> Configuring upload protocol...
> AVAILABLE: blackmagic, jlink, mbed, stlink
> CURRENT: upload_protocol = stlink
> Uploading .pio\build\nucleo_f446ze\firmware.elf
> xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> debug_level: 1
> 
> srst_only separate srst_nogate srst_open_drain connect_deassert_srst
> 
> **target halted due to debug-request, current mode: Thread** 
> xPSR: 0x01000000 pc: 0x08000abc msp: 0x20020000
> ** Programming Started **
> ** Programming Finished **
> ** Verify Started **
> ** Verified OK **
> ** Resetting Target **
> shutdown command invoked

Details:
I have windows 10.
I use the framework stm32cube.
I've tried to flash it with upload_protocol = mbed, but it won't blink.
The board does work in ubuntu with a different project using framework mbed.

valeros, 2020-03-25:
Hi @ChrisIdema ! Are you sure you're using the correct pin? Looks like the LEDs are connected to PB0, PB7, PB14 pins, so you need to modify the code accordingly.

ChrisIdema, 2020-03-26:
@valeros I tried the different pin, but it doesn't work.

valeros, 2020-03-26:
This board has a built-in debugger, could you please try to start a debugging session to find out what exactly is wrong?

ChrisIdema, 2020-03-26:
@valeros 
Ok I got it to blink. I had to enable the clock for port B. It blinks with both upload protocols. I was put on the wrong track by bright yellow text saying "target halted".
The only problem remaining is that when I enable freertos I cannot get it to blink.

valeros, 2020-03-26:
Can't help you here, looks like the issue is specific to your project only.

ChrisIdema, 2020-03-26:
How do I get freertos running? I didn't find an example yet.

valeros, 2020-03-26:
There is no native support for FreeRTOS, only some libraries for Arduino framework: https://platformio.org/lib/show/2093/STM32duino%20FreeRTOS

ChrisIdema, 2020-03-26:
I'm using this library: https://platformio.org/lib/show/6696/STM32Cube%20Middleware-FreeRTOS
According to FreeRTOS STM32 cube comes with FreeRTOS: https://www.freertos.org/FreeRTOS-Plus/BSP_Solutions/ST/STM32Cube.html

valeros, 2020-03-26:
As I said before, try to run debugger so you can figure out where exactly MCU stops working.
I'm closing this topic as it got off topic.

closed 2020-03-26 by valeros