platformio/platform-ststm32 #367: "fatal error: variant.h: No such file or directory" when using STM32Duino

HamidSaffari, 2020-04-03:
Hi that's my config:

[env:genericSTM32F103RC]
platform = ststm32
board = genericSTM32F103RC
framework = arduino
board_build.core = stm32 ;or STM32Duino

result:
C:\users\hamid\.platformio\packages\framework-arduinoststm32\system\STM32F1xx/stm32f1xx_hal_conf.h:4:10: fatal error: variant.h: No such file or directory

valeros, 2020-04-03:
Looks like you also need to specify the right variant:
```
board_build.variant = Generic_F103Rx
```

HamidSaffari, 2020-04-03:
No, still the same with  board_build.variant = Generic_F103Rx

HamidSaffari, 2020-04-03:
For this to working I had to includue special variant.h inside my main src code. like this:

#include <C:\Users\Hamid\.platformio\packages\framework-arduinoststm32\variants\Generic_F103Rx/variant.h>

And it does not automatically selected.

valeros, 2020-04-03:
Cannot reproduce, the following config works just fine:
```ini
[env:genericSTM32F103RC]
platform = ststm32
board = genericSTM32F103RC
framework = arduino
board_build.core = stm32
board_build.variant = Generic_F103Rx
```

HamidSaffari, 2020-04-03:
Oh, yes that
  board_build.variant = Generic_F103Rx
did the job. I just had to rebuild all again.
Thanks.

closed 2020-04-03 by valeros