platformio/platform-ststm32 #368: Package “framework-stm32cube/f4” is outdated

Remni1, 2020-04-05:
Hello, 
It looks like framework-stm32cube/f4 package is outdated, I explained it deeper here:
https://community.platformio.org/t/it-looks-like-package-framework-stm32cube-f4-is-outdated/12886

Is there a possibility to update this package?
Thank you in advance for your help.

JcBernack, 2020-05-11:
All the STM32 cube sources are outdated. Can we please get an update here?
As I can't find a repository for the framework-stm32cube package I assume it is auto-generated by condensing and zipping the official repos:
https://github.com/STMicroelectronics/STM32CubeF4 (and similar)
This makes it a bit more difficult to fork and update.

JcBernack, 2020-05-13:
I just created a repository to get the most recent sources into my build. Just add this to `platformio.ini`:

    platform_packages = 
        framework-stm32cube @ https://github.com/rocknitive/framework-stm32cube.git

Please keep in mind I just updated the source for the **L4** series. The others will be updated when I find the time or by PR.

daniel-starke, 2020-09-30:
All the STM32Cube sources are provided by STMicroelectronics on github in their latest version. E.g. https://github.com/STMicroelectronics/STM32CubeF1. Why aren't those used more or less directly? The files at JcBernack repository aren't up-to-date either (e.g. F1 is already released in version 1.8.1 instead of 1.8.0) and provide only a workaround for what PlatformIO provides out-of-the-box.

JcBernack, 2020-10-01:
I guess they are not used directly because they contain huge amounts of stuff which is not necessary just for compiling the project. There are tons of example projects, utilities and documentation which would only slow down the download and therefore the build, e.g the original L4 repo is 325MB. Have a look at the `.gitignore` file in my repo to see what I excluded. The repo also contains a script to automatically update all sources from the upstream repos by STM (which I just did).

daniel-starke, 2020-10-01:
Using the official sources directly does not necessary means that no filtering is applied before storing them locally. True, the data needed to download each provided release ZIP is much higher. But maybe there is a way to set up some Github triggers to auto-update such an intermediate repository like yours after each release by STMicroelectronics?

Thank you for updating your repository.

valeros, 2025-07-23:
Resolved in [v19.0.0](https://github.com/platformio/platform-ststm32/releases/tag/v19.0.0).

closed 2025-07-23 by valeros