platformio/platform-ststm32 #370: USB Keyboard - USB HID not enabled

ThomasHulme, 2020-04-09:
I'm trying to setup an STM32 Bluepill as a USB keyboard.
In the Arduino IDE the settings in Tools are:
[image]

In platformio.ini the settings I have are:
[image]

However I get the error:
#error "USB HID not enabled! Select 'HID' in the 'Tools->USB interface' menu."

I have the "PIO_FRAMEWORK_ARDUINO_ENABLE_HID" in the build flags.

Is there anything I'm missing??

jonasled, 2020-04-19:
I have the same problem, so I think this is a bug.

maxgerhardt, 2020-05-11:
> I have the "PIO_FRAMEWORK_ARDUINO_ENABLE_HID" in the build flags.
As you can see [in the docs](https://docs.platformio.org/en/latest/platforms/ststm32.html#stm32duino-configuration-system) this is a configuration for the STM32Duino core. But you have defined `genericSTM32F103C8` as the `board` in the `platformio.ini`, so it will use 

https://github.com/platformio/platform-ststm32/blob/develop/boards/genericSTM32F103C8.json#L3

the maple core (`framework-arduinoststm32-maple`) and not STM32Duino core. 

Then you also cross-mix libraries for STM32Duino into this build with the `lib_extra_dirs` command, which doesn't look healthy. 

So which exact core do you want to use?

jonasled, 2020-05-11:
I tried it also with `bluepill_f103c8` as board, which uses `stm32` as core and there the error also occurs.

jonasled, 2020-05-11:
My platformio.ini:
```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
build_flags = -D PIO_FRAMEWORK_ARDUINO_ENABLE_HID
```

maxgerhardt, 2020-05-11:
The framework needs additional configuration, see 

https://github.com/stm32duino/Arduino_Core_STM32/blob/3a9c71c6347c32e47ab3437a7713c96388d60605/libraries/Keyboard/src/Keyboard.h#L27-L29 

By using additional flags listed in https://community.platformio.org/t/difficulty-with-getting-usb-serial-usb-cdc-working/7501/6?u=maxgerhardt I can get an example to compile.

```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
build_flags = 
	-D PIO_FRAMEWORK_ARDUINO_ENABLE_HID
	-D USBCON
	-D USBD_VID=0x0483
	-D USB_MANUFACTURER="Unknown"
	-D USB_PRODUCT="\"BLUEPILL_F103C8\""
	-D HAL_PCD_MODULE_ENABLED
```

```cpp
#include <Arduino.h>
#include <Keyboard.h>

void setup() {
    Serial.begin(9600);
    Keyboard.begin();
}
void loop() {
	Keyboard.write((uint8_t)'A');
    Serial.println("Loop");
    delay(500);
}
```