platformio/platform-ststm32 #371: STM32Duino FreeRTOS 10.0.1 with PlatformIO Core 4.3.1 Build fail

piyushverma, 2020-04-11:
Hello,

Default example code of STM32Duino FreeRTOS build faild.
/home/piyush/.platformio/lib/STM32duino FreeRTOS_ID2093@9.0.4/src/portmacro.h:12:20: error: no "__CORTEX_M" definition
 #pragma GCC error  "no \"__CORTEX_M\" definition"
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio/build/genericSTM32F103CB/lib0d5/STM32duino FreeRTOS_ID2093@9.0.4/STM32FreeRTOS.c.o] Error 1
*** [.pio/build/genericSTM32F103CB/lib0d5/STM32duino FreeRTOS_ID2093@9.0.4/FreeRTOS/Source/timers.c.o] Error 1

Here is my platformio.ini
[env:genericSTM32F103CB]
platform = ststm32
board = genericSTM32F103CB
framework = arduino
lib_deps = STM32duino FreeRTOS

Any suggestion ?

valeros, 2020-04-13:
Hi @piyushverma ! Maybe this library doesn't support Maple core. Could you please try to compile your project using another board? For example `nucleo_f401re`?

fpistm, 2020-04-13:
Hi, @valeros said the right thing. This library is for the [STM32 core](https://github.com/stm32duino/Arduino_Core_STM32) which is based on CMSIS from arm which defined the cortex used.

valeros, 2020-06-24:
FreeRTOS_ID2093 cannot be used with maple core.

closed 2020-06-24 by valeros