platformio/platform-ststm32 #372: Multipel serial support on Blue Pill not work?

JoergSH, 2020-04-11:
I want to use 2 serial connection but don't work

```
Serial.begin(115200);
Serial1.begin(9600);
```

Serial1.begin has output on the same serial pins  as serial.

Any suggestions?

regards

Joerg

fpistm, 2020-04-14:
If you used the STM32 core, then Serial is the generic instance name and is linked to the default Serialx instance. In case of the BP, this is Serial1, so this is normal.
If you enable the USB CDC, the Serial will be SerialUSB and Serial1 is available.
You can also instantiate other Serial using HardwareSerial constructor:
https://github.com/stm32duino/wiki/wiki/API#hardwareserial

lhartmann, 2020-05-07:
I just can't get the SerialUSB to work. DFU bootloader works OK, but does not enumerate after my firmware uploads. An example would be highly appreciated.