platformio/platform-ststm32 #373: stm32f103xe.ld is missing

SuperBoss9, 2020-04-14:
File stm32f103xe.ld is missing in 6.1.0
That file is used by:
genericSTM32F103RE
genericSTM32F103VE
genericSTM32F103ZE
maple_ret6.json

No success compilation due to this

valeros, 2020-04-14:
Hi @SuperBoss9 ! What framework are you using?

SuperBoss9, 2020-04-14:
Hi @valeros, STMDuino (Arduino clone).

valeros, 2020-04-14:
`genericSTM32F*` boards are meant to be used with `maple` core, however I'm not able to reproduce the issue even if I select `STMDuino` core manually. If I'm not mistaken, `STMDuino` has its own linker scripts for each variant. What is your `platformio.ini`? Do you use custom a linker script?

SuperBoss9, 2020-04-14:
STMDuino uses STM32Core not maple. Actually it is possible to select appropriate core for most of the boards included F*.
My ini file is:

board = genericSTM32F103RE-2
platform = ststm32
framework = arduino
board_build.core = stm32 
debug_tool = stlink
build_flags = -DHAL_DAC_MODULE_ENABLED
              -fexceptions

And I use custom board definition and it has its own .ld stored at the variants folder of my Project. But it seems PIO tries to use stm32f103xe.ld that is obviously missed.

The whole error message is:
.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file stm32f103xe.ld: No such file or directory

valeros, 2020-04-14:
What is your board manifest? Is there a global linker script defined? Take a look at a recently updated config for [genericSTM32F103RE](https://github.com/platformio/platform-ststm32/blob/develop/boards/genericSTM32F103RE.json).

SuperBoss9, 2020-04-14:
Here is my board definition:
  "build": {
    "core": "stm32",
    "cpu": "cortex-m3",
    "extra_flags": "-DSTM32F103xE -DSTM32F1",
    "f_cpu": "72000000L",
    "hwids": [
      [
        "0x1EAF",
        "0x0003"
      ],
      [
        "0x1EAF",
        "0x0004"
      ]
    ],
    "ldscript": "stm32f103xe.ld",
    "mcu": "stm32f103ret6",
    "variants_dir": "variants",
    "variant": "Generic_F103Rx"	
  },
  "debug": {
    "jlink_device": "STM32F103RE",
    "openocd_target": "stm32f1x",
    "svd_path": "STM32F103xx.svd"
  },

As I see both of them uses "ldscript": "stm32f103xe.ld" script for linker. But where is it?

valeros, 2020-04-14:
Just delete `"ldscript": "stm32f103xe.ld",` line and it should work.

SuperBoss9, 2020-04-14:
I have to confirm that this helped. Thank you very much.
Also noticed that the core is updated to 1.8.0.0 version.

closed 2020-04-14 by SuperBoss9