platformio/platform-ststm32 #377: [Feature Request] CMSIS Framework for STM32F0xx

Schrolli91, 2020-04-29:
Hey there,

is there any reason, why PIO has no CMSIS Framework for the STM32F0xx-series?
I can only find the following CMSIS headers from ST in `....platformio\packages\framework-cmsis\variants\ststm32` 
But they provide also CMSIS Headers for the F0-series.

[image]

Maybe its possible to bring these heders into PIO.

Schrolli91, 2020-04-29:
@ivankravets  thanks for transfer to the right repo - sorry for opening in the wrong.
maybe thats a possible duplicate with #26 ?

ivankravets, 2020-04-29:
Yes,it's duplicate https://github.com/platformio/platform-ststm32/issues/26

We will back soon to https://github.com/platformio/platform-ststm32/issues/26

closed 2020-04-29 by ivankravets