platformio/platform-ststm32 #382: Project Wizard - Problem Win10 VSCode in WSL

ernstgross, 2020-05-15:
I'm using Win10 with WSL. The Project Wizard has a problem when VSCode is running in Remote WSL mode. It seems that the VSCode plugin uses the '\' backslash directory separator from Win10, but in this situation it needs to use a unix  style slash '/'  

[image]

This is the generated folder:

```
ernst@XXXX:~/Documents/PlatformIO$ ls -l
total 0
drwxrwxrwx 1 ernst ernst 4096 May 14 21:30  Projects
drwxrwxrwx 1 ernst ernst 4096 May 15 13:47 'Projects\BlinkyF723Disco'

ernst@XXXX:~/Documents/PlatformIO$ mv Projects\\BlinkyF723Disco/ Projects/BlinkyF723Disco

ernst@XXXX:~/Documents/PlatformIO$ ls -l
drwxrwxrwx 1 ernst ernst 4096 May 15 13:51 Projects

ernst@XXXX:~/Documents/PlatformIO$ ls -l Projects
drwxrwxrwx 1 ernst ernst 4096 May 15 13:47 BlinkyF723Disco
```

jniclas, 2020-10-24:
I have the same problem. Have you found a solution?

Querela, 2020-11-08:
Still a problem for me ...
I can't seem to load any existing project even after renaming the created ones.