platformio/platform-ststm32 #389: using upstream stm32 arduino core

stas2z, 2020-05-19:
while platformio still use 1.8.0 core version im trying to handle how to use that core from upstream

my test project platformio.ini
```
[env:genericSTM32F103RE]
platform = ststm32
board = genericSTM32F103RE
framework = arduino
platform_packages =
     framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32

board_build.core = stm32
board_build.variant = Generic_F103Rx
```
arduino core already contains tools/platformio-build.py for pio, but doesn't contains CMSIS folder which is required to build core files, it's packaged separately here: https://github.com/stm32duino/ArduinoModule-CMSIS

so upstream core will not compile as platformio-build.py checking existence of CMSIS folder inside the package

ive made a workaround to fetch and unpack CMSIS module at first use of the core (if CMSIS folder is not there)

```
diff --git a/tools/platformio-build.py b/tools/platformio-build.py
index 68b0dfc..8aa0e29 100644
--- a/tools/platformio-build.py
+++ b/tools/platformio-build.py
@@ -22,7 +22,7 @@ kinds of creative coding, interactive objects, spaces or physical experiences.
 https://github.com/stm32duino/Arduino_Core_STM32
 """
 
-
+import os
 from os.path import isfile, isdir, join
 
 from SCons.Script import DefaultEnvironment
@@ -34,8 +34,8 @@ board = env.BoardConfig()
 FRAMEWORK_DIR = platform.get_package_dir("framework-arduinoststm32")
 CMSIS_DIR = join(platform.get_package_dir("framework-arduinoststm32"), "CMSIS", "CMSIS")
 assert isdir(FRAMEWORK_DIR)
-assert isdir(CMSIS_DIR)
-
+if not isdir(CMSIS_DIR):
+   os.system("cd %s && wget -qO- %s | tar -xjf -" %(FRAMEWORK_DIR, "https://github.com/stm32duino/ArduinoModule-CMSIS/releases/download/5.5.1/CMSIS-5.5.1.tar.bz2"))
 
 mcu = env.BoardConfig().get("build.mcu", "")
 board_name = env.subst("$BOARD")
```

with this workaround ive built my test project successfully

sjasonsmith, 2020-05-31:
I have also been trying to figure out how to consume the 1.9 framework from within PlatformIO. As @stas2z said it can be used if you copy in the CMSIS file, but this isn't a very good option for other users to consume.

psobiech, 2020-06-24:
Instead of using @stas2z solution, I also experimented with using `framework-cmsis` and pointing platformio-build.py to `framework-cmsis` folder, but there were linker issues because `/DSP/Lib/GCC` does not exist so it couldn't find the libraries. 

For it to work with framework-cmsis, the following patch needs to be applied (because the Lib folder has different location):
```diff --git a/tools/platformio-build.py b/tools/platformio-build.py
index 68b0dfc..d84a5aa 100644
--- a/tools/platformio-build.py
+++ b/tools/platformio-build.py
@@ -32,7 +32,7 @@ platform = env.PioPlatform()
 board = env.BoardConfig()
 
 FRAMEWORK_DIR = platform.get_package_dir("framework-arduinoststm32")
-CMSIS_DIR = join(platform.get_package_dir("framework-arduinoststm32"), "CMSIS", "CMSIS")
+CMSIS_DIR = join(platform.get_package_dir("framework-cmsis"), "CMSIS")
 assert isdir(FRAMEWORK_DIR)
 assert isdir(CMSIS_DIR)
 
@@ -238,7 +238,7 @@ env.Append(
         "gcc",
         "stdc++",
     ],
-    LIBPATH=[variant_dir, join(CMSIS_DIR, "DSP", "Lib", "GCC")],
+    LIBPATH=[variant_dir, join(CMSIS_DIR, "Lib", "GCC")],
 )
 
 env.ProcessFlags(board.get("build.framework_extra_flags.arduino", ""))
```

Otherwise, with CMSIS provided a simple sketch seems to be working just fine (with both methods/cmsis versions). 

Additionally, even when providing CMSIS as @stas2z suggests and using CDC, we need to provide a `USBD_PID` define, with the previous version this define was ignored/not necessary. 

For reference, error when not providing `USBD_PID`:
```
error: 'USBD_PID' undeclared here (not in a function); did you mean 'USBD_VID'?
  179 |   LOBYTE(USBD_PID),           /* idProduct */
```

My `platformio.ini` for reference:
```
[env:genericSTM32F103ZE]
platform = https://github.com/platformio/platform-ststm32.git
platform_packages = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32

framework = arduino

;framework =
;    arduino
;    cmsis ; only used to fetch framework-cmsis

board = genericSTM32F103ZE
board_build.core = stm32
board_build.variant = Generic_F103Zx

build_flags =
    -D PIO_FRAMEWORK_ARDUINO_SERIAL_WITHOUT_GENERIC
    -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
    -D USBCON
    -D USBD_USE_CDC
    -D USBD_VID=0x0483
    -D USBD_PID=0x5740
    -D USB_MANUFACTURER="Unknown"
    -D USB_PRODUCT="\"STM32F103ZE\""
    -D HAL_PCD_MODULE_ENABLED

upload_protocol = dfu
upload_port = /dev/ttyACM0
```

maxgerhardt, 2020-06-25:
Resolved by official release https://github.com/platformio/platform-ststm32/releases/tag/v7.0.0.