platformio/platform-ststm32 #395: stm32duino and board_build.core

jorgegarciadev, 2020-05-27:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] **PlatformIO Core**.

      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 20.04

**PlatformIO Version** (`platformio --version`): 4.3.3

### Description of problem
I've found that PlatformIO install arduinoststm32-maple core instead of arduinoststm32 core as default. In the [documentation](https://docs.platformio.org/en/latest/platforms/ststm32.html#switching-between-arduino-cores) says that the default one is arduinoststm32, and that it can be overridden using `board_build.core` to install arduinoststm32-maple. I don't know if this is a bug or if the documentation is wrong or outdated.

#### Steps to Reproduce

1. Create a new project using ststm32.
2. Build the project and install the platform, 'arduinoststm32-maple' is installed. 
3. Create a new project using `ststm32` and set `board_build.core` to `ststm32`.
2. Build the project and install the platform, 'arduinoststm32' is installed. 

### Actual Results
When creating a new `ststm32` project installs arduinoststm32-maple platform, contradicting what the documentation states.

### Expected Results
When creating a new `ststm32` project installs arduinoststm32 platform, as the documentation says.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:blackboard]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
upload_protocol = stlink
board_build.core = ststm32
lib_deps =
  Wire
  Adafruit ST7735 and ST7789 Library
  Adafruit GFX Library

```

valeros, 2020-05-27:
That's correct behavior, `maple` core is default choice for some boards.

closed 2020-05-27 by valeros

jorgegarciadev, 2020-05-27:
Then please update the documentation stating that, and include the `board_build.core = ststm32` override in the "Switching between Arduino cores" section.