platformio/platform-ststm32 #396: Arduino IWatchdog error

pilotak, 2020-05-28:
compiling https://github.com/stm32duino/Arduino_Core_STM32/blob/1f6168462ce2ef0e88578773161956387d938c19/libraries/IWatchdog/examples/IWDG_Button/IWDG_Button.ino gives me an error.

```
fatal error: IWatchdog.h: No such file or directory

*******************************************************************
* Looking for IWatchdog.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:IWatchdog.h"
* Web  > https://platformio.org/lib/search?query=header:IWatchdog.h
*
*******************************************************************

     #include <IWatchdog.h>
              ^~~~~~~~~~~~~
compilation terminated.
```
> It's interesting that servo library compiles files - it's in the same dir as watchdog

PlatformIO, version 4.3.4
PLATFORM: ST STM32 6.1.0 > STM32F103C8 (20k RAM. 64k Flash)
```ini
[env:test]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
upload_protocol = stlink
```

maxgerhardt, 2020-06-10:
Adding `lib_deps = IWatchdog` in the `platformio.ini` makes no difference?

pilotak, 2020-06-12:
@maxgerhardt unfortunately even with added as dependency
`fatal error: IWatchdog.h: No such file or directory`

pilotak, 2020-08-16:
seems to be fixed in version 8.0.0

closed 2020-08-16 by pilotak