platformio/platform-ststm32 #397: Upload tool hid-flash not found

albertskog, 2020-05-28:
Hi, I'm trying to use the `hid` bootloader and upload option but I'm getting an error saying `hid-flash` is not found:

```
$ pio run -e usb -t upload -v
Processing usb (upload_protocol: hid; upload_port: ttyACM0; platform: ststm32@6.1.0; board: bluepill_f103c8; framework: arduino; build_type: debug; build_flags: -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC, -D USBCON, -D USBD_VID=0, -D USB_PRODUCT="\"BLUEPILL_F103C8\"")
-----------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 6.1.0 > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10800.200207 (1.8.0) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.190707 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
MethodWrapper(["checkprogsize"], [".pio/build/usb/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.6% (used 3812 bytes from 20480 bytes)
Flash: [====      ]  37.0% (used 24232 bytes from 65536 bytes)
.pio/build/usb/firmware.elf  :
section               size        addr
.isr_vector            268   134219776
.text                22288   134220044
.rodata               1428   134242332
.init_array             24   134243760
.fini_array              8   134243784
.data                  516   536870912
.bss                  3296   536871428
._user_heap_stack     1540   536874724
.ARM.attributes         41           0
.debug_info         454509           0
.debug_abbrev        53270           0
.debug_aranges        7848           0
.debug_line         144543           0
.debug_str           56529           0
.comment               102           0
.debug_ranges         8712           0
.debug_frame         21408           0
.debug_loc          104089           0
Total               880419
<lambda>(["upload"], [".pio/build/usb/firmware.bin"])
AVAILABLE: blackmagic, dfu, hid, jlink, mbed, stlink
CURRENT: upload_protocol = hid
MethodWrapper(["upload"], [".pio/build/usb/firmware.bin"])
Use manually specified: ttyACM0
hid-flash ".pio/build/usb/firmware.bin" "ttyACM0"
sh: 1: hid-flash: not found
*** [upload] Error 127
```

Here is my Platformio.ini:
```
[env]
platform = ststm32@6.1.0
board = bluepill_f103c8
framework = arduino
build_flags =
    -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
    -D USBCON
    -D USBD_VID=0
    -D USB_PRODUCT="\"BLUEPILL_F103C8\""

[env:usb]
upload_protocol = hid
upload_port = ttyACM0
```
Some observations:
* Looking at the [Platformio docs](https://docs.platformio.org/en/latest/boards/ststm32/bluepill_f103c8.html#uploading), `hid` is not mentioned as one of the upload options, probably because it was added to Arduino_Core_STM32 fairly recently.
* There is also no mention of `hid-flash` in [platform.json](https://github.com/platformio/platform-ststm32/blob/develop/platform.json), or anywhere else that I can find.
* Thirdly, there is no `hid-flash` in `~/.platformio/packages/tool-stm32duino`
* In Arduino_Core_STM32, `hid-flash` was added in [1.3.0](https://github.com/stm32duino/Arduino_Tools/releases/tag/1.3.0)
* In the [latest release (6.1.0) of platform-ststm32](https://github.com/platformio/platform-ststm32/releases/tag/v6.1.0), Arduino_Core_STM32 is on 1.8.0 so `hid-flash` _should_ be there, right?

I have not been this deep into the internals of Platformio before but happy to learn, any pointers on what to do?

KenjutsuGH, 2020-05-30:
I am using macOS, and the following worked for me:

1. I downloaded **hid-flash** from [STM32_HID_Bootloader](https://github.com/Serasidis/STM32_HID_Bootloader)
2. I then copied the appropriate version for my operating system to `~/.platformio/packages/tool-stm32duino/`

That did the trick for me

albertskog, 2020-06-02:
I confirmed this fixes the issue for me as well. Looking a bit deeper, it seems the root cause is that `hid-flash` is missing from here: https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/tools/linux64

I have opened an issue over there instead: https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/794

closed 2020-06-02 by albertskog