platformio/platform-ststm32 #399: Support for Electrosmith Daisy

wraybowling, 2020-06-08:
What kind of issue is this?

- **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform
------------------------------------------------------------------

### Additional info

ElectroSmith Daisy is based on STM32H750IBK6, and has details on the toolchain here: https://github.com/electro-smith/DaisyWiki/wiki/1.-Getting-Started#1-Install-the-Toolchain

library here: https://github.com/electro-smith/libDaisy/tree/0c753cb347547d3483146439d4e9e772b44c1fa8

examples here: https://github.com/electro-smith/DaisyExamples

wraybowling, 2020-06-08:
cc: @electro-smith Y'all may have all the expertise to make this work already.

flavius-popan, 2020-06-23:
Came here to request this as well, bump!!!

valeros, 2020-06-25:
Resolved in https://github.com/platformio/platform-ststm32/commit/53cf020444d2cc0d52b1fcee80480f3e71be007b

closed 2020-06-25 by valeros

wraybowling, 2020-06-25:
Wooo! 🥳