platformio/platform-ststm32 #400: Bluepill can't start mbed-os FW due to sleep timer

maxgerhardt, 2020-06-10:
Refer https://community.platformio.org/t/stm32f103c8-mbed-support/12302/

Basically the `framework-mbed` package has a `targets.json` with an entry

```
    "BLUEPILL_F103C8": {
        "inherits": ["FAMILY_STM32"],
        "core": "Cortex-M3",
        "default_toolchain": "GCC_ARM",
        "extra_labels_add": ["STM32F1", "STM32F103C8"],
        "supported_toolchains": ["GCC_ARM"],
        "device_has_add": [
            "CAN",
            "SERIAL_ASYNCH",
            "FLASH"
        ],
        "device_has_remove": ["STDIO_MESSAGES", "LPTICKER"]
    },
```
which the original targets.json (https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json) doesn't have.

However, the target definition is wrong -- Although `LPTICKER` is removed, the setting `tickless-from-us-ticker: false` is still set. This will cause mbed-os recognize that it's configured to **not use** a µS-Timer, i.e. an low-power timer, which it however doesn't have an hardware. An `MBED_ASSERT()` is triggered for that when calling `thread_sleep_for()` and the firmware hangs up.

The `targets.json` must correct so that it reads

```
    "BLUEPILL_F103C8": {
        "inherits": ["FAMILY_STM32"],
        "core": "Cortex-M3",
        "default_toolchain": "GCC_ARM",
        "extra_labels_add": ["STM32F1", "STM32F103C8"],
        "supported_toolchains": ["GCC_ARM"],
        "device_has_add": [
            "CAN",
            "SERIAL_ASYNCH",
            "FLASH"
        ],
        "device_has_remove": ["STDIO_MESSAGES", "LPTICKER"],
	"overrides": {
            "tickless-from-us-ticker": true
        }
    },
```

i.e., tickless from us-ticker is set to **true** to use the only available type of timers. 

Other custom targets may have the same error?

scarleast, 2020-06-11:
I now have a STM32F103C8T6 development board that I made myself, and I hope to use Mbed for some development.

I used Mbed Studio to develop before, the target is NUCLEO_F103RB, it really works fine.

But with Platformio, choose NUCLEO_F103RB for the target , and this problem(thread_sleep_for() don‘t work) still exists.

Now, since Platformio can choose bluepill to develop with Mbed, there is no need to choose NUCLEO_F103RB, but I suggest to check whether the thread_sleep_for function of NUCLEO_F103RB is normal.

Finally, thank PlatformIO maintainers for their support of bluepill.

leonty, 2021-01-11:
I have come across this problem with an stm32 min dev board clone which configured as
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = mbed

maxgerhardt, 2021-01-12:
@valeros can we fix the `targets.json` file in the framework-mbed package that is used for the bluepill here?