platformio/platform-ststm32 #402: Enable C++ exception handling?

Crazor, 2020-06-22:
How do I enable C++ exception handling on STM32?

I've already added `build_unflags = -fno-exceptions` and `build_flags = -fexceptions` to `platformio.ini`, but I can't catch any. They get thrown, the debugger shows the backtrace contains `__cxa_throw`, `std::terminate`, `abort` and `kill`. But even a blanket catch does nothing:

```
  try {
    throw(std::exception());
  } catch (std::exception& e) {
    std::cout << "Exception caught." << std::endl;
  }
```

My compiler flags (scraped directly from the compilation output):
`-std=gnu++17 -fexceptions -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m4 -nostdlib -DPLATFORMIO=40304 -DSTM32F407xx -DARDUINO_STM32GenericF407VET6 -DSTM32F4 -DDEBUG -DUSE_HAL_DRIVER -DF_CPU=168000000L`

My linker flags:
`-Wl,-u_printf_float -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m4 --specs=nano.specs --specs=rdimon.specs  -Wl,--start-group -lc -lgcc -lm -lstdc++ -lrdimon -Wl,--end-group`

Crazor, 2020-06-22:
Ah well, I figured it out. I had to add `--specs=nano.specs` to build_unflags as well. Code size went up by about 150kb, but the F407 can handle that (1M of flash).

valeros, 2020-06-22:
Hi @Crazor ! What framework do you use?

Crazor, 2020-06-22:
I'm using `stm32cube` with the project generated by `stm32pio`.

valeros, 2020-06-22:
`--specs=nano.specs` is the default choice for the most low-level frameworks in this platform (including `stm32cube`) and as you correctly stated the exceptions don't work with this flag because newlib-nano version of the standard library is built without exception support. I can only confirm that your solution is the right way of enabling exceptions at the moment.

closed 2020-06-22 by valeros

Crazor, 2020-06-22:
That's about how I had it in my head as well. However, as I'm no expert with newlib, can you please point me at some description of what different values for `--specs` are available and what is different between those? Is this dependent on PlatformIOs own build of newlib-nano?

valeros, 2020-06-22:
Unfortunately, there is no comprehensive documentation on this topic. The list of available `--specs` options mostly depends on the toolchain that's used to compile the code. In the most cases we rely on the official toolchains from ARM https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads