platformio/platform-ststm32 #403: When will framework-arduinoststm32 4.10800.200207 (1.8.0) be updated to 1.9?

exothink, 2020-06-23:
1.9 has been released over a month ago. I'm trying to use hardware timer, but examples no longer run on 1.8

maxgerhardt, 2020-06-23:
Duplicate of https://github.com/platformio/platform-ststm32/issues/389 and https://github.com/platformio/platform-ststm32/issues/385.

As you can see in https://bintray.com/platformio/dl-packages/framework-arduinoststm32#files, the version `framework-arduinoststm32-4.10900.200602.tar.gz` already exists for 1.9.0, so it can also be manually enabled 

```
platform_packages =
  framework-arduinoespressif8266 @ 4.10900.200602
```
([docs](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages)) but will surely be merged shortly.

psobiech, 2020-06-24:
@exothink You could try your luck with the following, because for me the `arduinoststm32-4.10900.200602` version did not work due to missing CMSIS folder.

```ini
platform = https://github.com/psobiech/platform-ststm32.git
platform_packages = framework-arduinoststm32@https://github.com/psobiech/Arduino_Core_STM32.git
```

It is a fork of main repository, only with changes I mentioned in my comment here: https://github.com/platformio/platform-ststm32/issues/389#issuecomment-648766175

maxgerhardt, 2020-06-25:
Resolved by official release https://github.com/platformio/platform-ststm32/releases/tag/v7.0.0.

closed 2020-06-25 by valeros