platformio/platform-ststm32 #404: Toolchain not available for linux_arm7l (Raspberry Pi)

juleq, 2020-06-26:
Just updated pio. It uses 7.0.0 now and that cannot satisfy the required 1.9.x toolchain for a linux_arm7l arch (Raspberry Pi host, STM32 target).

juleq, 2020-06-26:
Was probably me. Was on a wrong branch. The right one for me required <6.2.0 and that is available.

juleq, 2020-06-27:
Reopening because I have another platform (lpc1768) that requires a >= 1.6.x toolchain, pio resolves to 1.9.x and that does not seem to be available for linux_arm7l. Since I am not sure where to raise this issue, I thought I just ask for 1.9.x toolchain support for Raspberry hosts.

derneue, 2020-07-03:
Hi,

I am also facing this issue. Anything new here?
`
Processing bluepill_f103c8_128k (platform: ststm32; board: bluepill_f103c8_128k; framework: arduino)
--------------------------------------------------------------------------------
PackageManager: Installing toolchain-gccarmnoneeabi @ ~1.90201.0
Error: Could not find a version that satisfies the requirement '~1.90201.0' for your system 'linux_armv7l'
`

valeros, 2020-08-13:
GCC v9.3.1 should be available for ARM platforms, please try the following configuration:

```
[env:override_default_toolchain]
platform = ststm32
board = bluepill_f103c8_128k
framework = arduino
platform_packages =
  toolchain-gccarmnoneeabi@~1.90301.0
```

terryb58, 2021-07-07:
It's 2021 and I'm still encountering this.  Even with valero's suggestion above, I still get the error:
"Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ ~1.90201.0' requirements for your system 'linux_armv7l'"
Is there a fix?

adamkrawczyk, 2021-07-16:
Also got this problem.

vodka-bears, 2022-01-06:
It's 2022 and there's still the problem.

isarrider, 2022-03-15:
I have the same issue, ut with 64bit:

`Processing STM32F103RET6_creality (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing ststm32 @ ~12.1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: ststm32 @ 12.1.1 has been installed!
The platform 'ststm32@~12.1' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.90201.0
Error: Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ ~1.90201.0' requirements for your system 'linux_aarch64'
`

dev-null2019, 2023-09-25:
It's 2023 and I'm still encountering this!

pvbaudin, 2024-02-02:
> GCC v9.3.1 should be available for ARM platforms, please try the following configuration:
> 
> ```
> [env:override_default_toolchain]
> platform = ststm32
> board = bluepill_f103c8_128k
> framework = arduino
> platform_packages =
>   toolchain-gccarmnoneeabi@~1.90301.0
> ```

This worked for me, chiming in from 2024