platformio/platform-ststm32 #406: add mbed and stm32cube support for stm32WB55 and p_nucleo stm32WB55

weatherhead99, 2020-07-06:
Hi,
I'm fairly new to platformio (though not to STM32 and mbed development), and I notice that the STM32WB55 board is supported both by mbed OS (https://os.mbed.com/platforms/ST-Nucleo-WB55RG/), and by stm32cube software (https://github.com/STMicroelectronics/STM32CubeWB).

I am interested in using platformio on a project targeting both the nucleo P board and custom hardware, however when configuring "framework=mbed" or "framework=stm32cube" I get error messages, because the only framework defined in the target is "arduino". I am unsure what work would be required to make these work (I am particularly interested in getting the mbed framework to work), and would be willing to put that effort in if someone could point me as to where to start.

Thanks

teo3n, 2022-03-22:
It has been almost two years now, are there any news on this one especially when it comes to the stm32cube support for the WB -series?

bwssytems, 2022-09-25:
Yep, this is pretty bad as I would like to use platformio with mbed on a stm32wb55rgv6. It seems it needs to just have the correct src include id for mbed as it is:

Failed to extract configuration for NUCLEO_WB55RG_P.
It might not be supported in the this Mbed release.

and should be something like NUCLEO_WB55RG as that is what is in the targets files for framework-mbed in platformio