platformio/platform-ststm32 #407: Travis win build fails afer zephyr upgrade in 7.1.0

puzrin, 2020-07-07:
https://travis-ci.org/github/lvgl/lv_platformio/jobs/705753584

```
PlatformManager: Installing ststm32

Downloading...

Unpacking...

Warning! Package Mirror: [WinError 206] The filename or extension is too long: 'C:\\Users\\travis\\.platformio\\platforms\\_tmp_installing-srig6fs1-package\\examples\\zephyr-cpp-synchronization\\.pio\\build\\disco_l475vg_iot01a\\..__framework-zephyr-20300-hal-stm32__stm32cube\\modules\\framework-zephyr-20300-hal-stm32\\stm32cube\\stm32l4xx\\drivers'

Looking for another mirror...

Downloading...

Unpacking...

Warning! Package Mirror: [WinError 206] The filename or extension is too long: 'C:\\Users\\travis\\.platformio\\platforms\\_tmp_installing-zysd812a-package\\examples\\zephyr-cpp-synchronization\\.pio\\build\\disco_l475vg_iot01a\\..__framework-zephyr-20300-hal-stm32__stm32cube\\modules\\framework-zephyr-20300-hal-stm32\\stm32cube\\stm32l4xx\\drivers'

Looking for another mirror...

The command "platformio run -e stm32f429_disco" exited with 1.
```

Updated package has too long paths inside to unpack.

ivankravets, 2020-07-07:
Could you try to set https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_CORE_DIR to `C:\.pio` in your Travis.CI  config?

valeros, 2020-07-07:
@puzrin That issue should be fixed in the latest release, so `PLATFORMIO_CORE_DIR` is not necessary. Please try to restart your CI without `PLATFORMIO_CORE_DIR`.

puzrin, 2020-07-07:
https://travis-ci.org/github/lvgl/lv_platformio/jobs/705753584 now works "as is".

Thank you for awesome support!

closed 2020-07-07 by puzrin