platformio/platform-ststm32 #408: ststm32 falils to update to 7.1.1 because of a 404 status code

bibipoggio, 2020-07-07:
Platform ST STM32
--------
Updating ststm32                         @ 6.1.1          [7.1.1]
Uninstalling ststm32 @ 6.1.1:   [OK]
PlatformManager: Installing ststm32 @ 7.1.1
Warning! Package Mirror: Got an unrecognized status code '404' when downloaded https://dl.registry.platformio.org/download/platformio/platform/ststm32/7.1.1/ststm32-7.1.1.tar.gz
Looking for another mirror...
Warning! Package Mirror: Got an unrecognized status code '404' when downloaded https://dl.registry.platformio.org/download/platformio/platform/ststm32/7.1.1/ststm32-7.1.1.tar.gz
Looking for another mirror...
Error: Could not install 'ststm32' with version requirements '7.1.1' for your system 'windows_x86'.

ivankravets, 2020-07-07:
Please try again, should work now.

Sorry for the issue.

closed 2020-07-07 by ivankravets

bibipoggio, 2020-07-07:
thanks @ivankravets 
just a noob question. I'm trying to build an upgraded version of marlin, but everytime I try to do so the vs code downloads the 6.1.1 version and fails to build it. Is it expected? or should it use the new 7.1.1 version to build it?
[image]

puzrin, 2020-07-09:
@bibipoggio See https://github.com/MarlinFirmware/Marlin/commit/5513e67512481024e0d571d84487f7c636996803. All developers pin dependencies to guarantee stability. You certainly should not try to override that.

If you have build problems with concrete project - ask in Marlin repo or their forum first.