platformio/platform-ststm32 #418: OpenOCD w/ stm32f7508-disco fails

jcw, 2020-07-31:
(copied from forum post: <https://community.platformio.org/t/openocd-w-stm32f7508-disco-fails/15207>)

I'm using PIO 4.3.4, and using "stlink" as upload protocol from the command line.

OpenOCD version is 0.10.0+dev (2019-07-17-15:21), the failure I get is:
```text
~/.platformio/packages/tool-openocd/bin/openocd -d2 -s /Users/jcw/.platformio/packages/tool-openocd/scripts -f board/stm32f7discovery.cfg -c "program {.pio/build/disco750/firmware.elf}  verify reset; shutdown;"
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-15:21)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J31M21 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.239604
Warn : Silicon bug: single stepping may enter pending exception handler!
Info : stm32f7x.cpu: hardware has 8 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080007a0 msp: 0x20050000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
** Programming Started **
Info : device id = 0x10016449
Info : flash size = 64 kbytes
Assertion failed: (i < bank->num_sectors), function setup_sector, file /Users/ilg/Work/openocd-0.10.0-13/openocd.git/src/flash/nor/stm32f2x.c, line 901.
Abort trap: 6
```

The "stm32f2x.c" bit looks like a mistake to me, but I haven't investigated further.

With `upload_protocol = mbed`, it works fine, but that's a bit slower.

Also, with the (newer) OpenOCD version 0.10.0+dev-01243-ge41c0f49-dirty (2020-05-18-16:46), the upload works:
```
** Programming Started **
Info : device id = 0x10016449
Info : flash size = 64 kbytes
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
```

That last one is installed via homebrew, I'm on latest MacOS.

valeros, 2020-08-03:
The OpenOCD package has been updated, please run `pio update`

closed 2020-08-03 by valeros

jcw, 2020-08-03:
Confirmed. Fantastic, many thanks.