platformio/platform-ststm32 #420: Usb Serial broke after platform update

saloid, 2020-08-08:
I can't launch USB serial at current (latest) version of ststm32 platform (7.2.0). At previous version (something like 6.8.0) all was OK.
My [platformio.ini from previous](https://github.com/giver0/Rocket-logger/blob/d4e2ccd04af693715e040cecfa4fe082886902b8/Soft/Embedded/Main-program/platformio.ini) version don't works anymore (but it was working perfectly until update). I was getting error at compilation ([see here](https://github.com/giver0/Rocket-logger/runs/937568851#step:5:143)).
My current platformio.ini looks like [this](https://github.com/giver0/Rocket-logger/blob/9e27dc3ad328d2d5c534be35132514a981a4e359/Soft/Embedded/Main-program/platformio.ini) and I'm getting this in device manager:
[image]
I am doing this on my custom board with STM32F103RCT6, but USB part is same to bluepill:
[image]

saloid, 2020-08-09:
I have downloaded Arduino IDE and launched project wit USB Serial usage in it - there it works.
So I have checked which flags are used in compilation in Arduino IDE regarding to USb and here thay are:
```ini
-DUSBCON
-DUSBD_VID=0
-DUSBD_PID=0
-DHAL_PCD_MODULE_ENABLED
-DUSBD_USE_CDC
```
I have added them to platformio.ini and get USB working in Platformio.

So currently my question is: 
What are [this flags](https://docs.platformio.org/en/latest/platforms/ststm32.html#id5) for? Are they supposed to replace this arduino flags, which can change time to time? If so, they should be fixed, since I have never seen them working neither in previous stm32 core version nor in current.

valeros, 2020-08-10:
Hi @saloid ! What Arduino core do you use now? Starting with ststm32 `v7.0.0` the STM32Duino core is used by default for all boards (except Maple).

That flags were added with the intent to replace the Arduino flags. Could you please specify what flags are missing (besides VID, PID) when you use
```
-D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
-D PIO_FRAMEWORK_ARDUINO_NANOLIB_FLOAT_PRINTF
-D PIO_FRAMEWORK_ARDUINO_USB_HIGHSPEED_FULLMODE
```

Thanks!

saloid, 2020-08-10:
I'm using stm32duino 7.2.0. These `-DPIO_FRAMEWORK...` flags have no effect (at least in my case) if I use ONLY them - USB Serial not works. If I use flags extracted from Arduino IDE (`-DUSBCON...`) - USB serial works WITHOUT `-DPIO_FRAMEWORK...` flags. I have not tried to combine them.

saloid, 2020-08-10:
As I understood how it supposed to work: if define PIO flags, they should automatically trigger native Arduino flags usage at build (which can be different at various core versions, but PIO flags always stays same for compatibility). But it doesn't work that way.

valeros, 2020-08-10:
The build script for STM32Duino is bundled together with the package, so it's possible to have different USB flags. You can greatly help if you find what flags are missing exactly or you can upload somewhere two build logs:
 - log from PlatformIO with enabled PIO flags
 - log from the Arduino IDE with USB configuration what works with your board

saloid, 2020-08-10:
## PIO flags
Log from platformIO with enabled PIO flags can be seen [here](https://github.com/giver0/Rocket-logger/runs/961377679), from CI build.
Single line example from it:
```shell
arm-none-eabi-gcc -o .pio/build/genericSTM32F103RC/FrameworkArduinoVariant/PeripheralPins.c.o -c -std=gnu11 -Wall -Wextra -Os -mcpu=cortex-m3 -mthumb -ffunction-sections -fdata-sections -Wall -nostdlib --param max-inline-insns-single=500 -DPLATFORMIO=40304 -DSTM32F103xE -DSTM32F1 -DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC -DPIO_FRAMEWORK_ARDUINO_NANOLIB_FLOAT_PRINTF -DPIO_FRAMEWORK_ARDUINO_USB_HIGHSPEED_FULLMODE -DHAL_SD_MODULE_ENABLED -DSTM32F1xx -DARDUINO=10808 -DARDUINO_ARCH_STM32 -DARDUINO_GENERICSTM32F103RC -DBOARD_NAME=\"GENERICSTM32F103RC\" -DHAL_UART_MODULE_ENABLED -DUSBD_USE_CDC -DHAL_PCD_MODULE_ENABLED -DUSE_USB_HS -DUSE_USB_HS_IN_FS -Ilib/SD_driver -I/home/runner/.platformio/packages/framework-arduinoststm32/cores/arduino/avr -I/home/runner/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32 -I/home/runner/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/LL -I/home/runner/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/usb -I/home/runner/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/OpenAMP -I/home/runner/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/usb/hid -I/home/runner/.platformio/packages/framework-arduinoststm32/cores/arduino/stm32/usb/cdc -I/home/runner/.platformio/packages/framework-arduinoststm32/system/Drivers/STM32F1xx_HAL_Driver/Inc -I/home/runner/.platformio/packages/framework-arduinoststm32/system/Drivers/STM32F1xx_HAL_Driver/Src -I/home/runner/.platformio/packages/framework-arduinoststm32/system/STM32F1xx -I/home/runner/.platformio/packages/framework-arduinoststm32/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc -I/home/runner/.platformio/packages/framework-arduinoststm32/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src -I/home/runner/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP -I/home/runner/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP/open-amp/lib/include -I/home/runner/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP/libmetal/lib/include -I/home/runner/.platformio/packages/framework-arduinoststm32/system/Middlewares/OpenAMP/virtual_driver -I/home/runner/.platformio/packages/framework-cmsis/CMSIS/Core/Include -I/home/runner/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include -I/home/runner/.platformio/packages/framework-arduinoststm32/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc -I/home/runner/.platformio/packages/framework-cmsis/CMSIS/DSP/Include -I/home/runner/.platformio/packages/framework-arduinoststm32/cores/arduino -I/home/runner/.platformio/packages/framework-arduinoststm32/variants/Generic_F103Rx -I/home/runner/.platformio/packages/framework-arduinoststm32/variants/Generic_F103Rx /home/runner/.platformio/packages/framework-arduinoststm32/variants/Generic_F103Rx/PeripheralPins.c
```
Flags from build log about usb:
```
-DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC
-DPIO_FRAMEWORK_ARDUINO_NANOLIB_FLOAT_PRINTF
-DPIO_FRAMEWORK_ARDUINO_USB_HIGHSPEED_FULLMODE
-DUSBD_USE_CDC
-DHAL_PCD_MODULE_ENABLED
-DUSE_USB_HS
-DUSE_USB_HS_IN_FS
```

## Arduino IDE flags
I can't launch it now (I can do it tomorrow, I believe), but here are the flags (as I mentioned before), which I already extracted from Arduino IDE build log from latest stm32duino board package (1.9.0):
```
-DUSBCON
-DUSBD_VID=0
-DUSBD_PID=0
-DHAL_PCD_MODULE_ENABLED
-DUSBD_USE_CDC
```

valeros, 2020-08-10:
Thanks for the info, do you use the same settings in Arduino IDE regarding `nanolib` version and USB speed?
To me it seems that the main difference `-DUSBCON` flag.

saloid, 2020-08-10:
I need only basic USB Serial, so for me it doesn't matter which USB speed is used.
Settings in Arduino IDE:
[image]
As I understood, these settings should correspond only to `-DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC` flag in platformio

Example line from Arduino IDE compilation:
```
"C:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\xpack-arm-none-eabi-gcc\\9.2.1-1.1/bin/arm-none-eabi-gcc" -mcpu=cortex-m3 -mthumb "@C:\\Users\\Dmytro\\AppData\\Local\\Temp\\arduino_build_1970/sketch/build_opt.h" -c -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -MMD "-IC:\\Program Files (x86)\\Arduino\\examples\\01.Basics\\AnalogReadSerial" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/avr" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/LL" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/OpenAMP" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/hid" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino/stm32/usb/cdc" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32F1xx_HAL_Driver/Inc" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/STM32F1xx_HAL_Driver/Src" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/STM32F1xx" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/open-amp/lib/include" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/libmetal/lib/include" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Middlewares/OpenAMP/virtual_driver" -DSTM32F1xx -DARDUINO=10813 -DARDUINO_GENERIC_F103RC -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"GENERIC_F103RC\"" -DSTM32F103xE -DUSBCON -DUSBD_VID=0 -DUSBD_PID=0 -DHAL_PCD_MODULE_ENABLED -DUSBD_USE_CDC "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/Core/Include/" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\tools\\CMSIS\\5.5.1/CMSIS/DSP/Include" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino" "-IC:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\variants\\Generic_F103Rx" "C:\\Users\\Dmytro\\AppData\\Local\\Arduino15\\packages\\STM32\\hardware\\stm32\\1.9.0\\cores\\arduino\\stm32\\usb\\cdc\\usbd_cdc.c" -o "C:\\Users\\Dmytro\\AppData\\Local\\Temp\\arduino_build_1970\\core\\stm32\\usb\\cdc\\usbd_cdc.c.o"

```

saloid, 2020-08-11:
# My little research

## USB Serial/HID
<details>
  <summary>-DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC</summary>

PlatformIO (wrong)
```
-DUSBD_USE_CDC
-DHAL_PCD_MODULE_ENABLED
```

Arduino IDE (correct)
```
-DUSBCON
-DUSBD_VID=0
-DUSBD_PID=0
-DHAL_PCD_MODULE_ENABLED
-DUSBD_USE_CDC
```
[image]
</details>

<details>
  <summary>-DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC_WITHOUT_SERIAL</summary>

PlatformIO (not tested)
```
-DUSBD_USE_CDC
-DDISABLE_GENERIC_SERIALUSB
-DHAL_PCD_MODULE_ENABLED
```

Arduino IDE (not tested)
```
-DUSBCON
-DUSBD_VID=0
-DUSBD_PID=0
-DUSBD_USE_CDC
-DDISABLE_GENERIC_SERIALUSB
-DHAL_PCD_MODULE_ENABLED
```
[image]
</details>
 
<details>
  <summary>-DPIO_FRAMEWORK_ARDUINO_ENABLE_HID</summary>

PlatformIO (not tested)
```
-DUSBD_USE_HID_COMPOSITE
-DHAL_PCD_MODULE_ENABLED
```

Arduino IDE (not tested)
```
-DUSBCON
-DUSBD_VID=0
-DUSBD_PID=0
-DHAL_PCD_MODULE_ENABLED
-DUSBD_USE_HID_COMPOSITE
```
[image]
</details>

Looks same, just need to add `-DUSBCON -DUSBD_VID=0 -DUSBD_PID=0` flags at each option

## High/Full Speed 
<details>
  <summary>-DPIO_FRAMEWORK_ARDUINO_USB_HIGHSPEED and -DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC</summary>

PlatformIO (not tested)
```
-DUSBD_USE_CDC
-DHAL_PCD_MODULE_ENABLED
-DUSE_USB_HS
```

Arduino IDE (not tested)
```
-DUSBCON
-DUSBD_VID=0
-DUSBD_PID=0
-DUSBD_USE_CDC
-DHAL_PCD_MODULE_ENABLED
-DUSE_USB_HS
```
[image]
</details>

<details>
  <summary>-DPIO_FRAMEWORK_ARDUINO_USB_HIGHSPEED_FULLMODE and -DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC</summary>

PlatformIO (not tested)
```
-DUSBD_USE_CDC
-DHAL_PCD_MODULE_ENABLED
-DUSE_USB_HS
-DUSE_USB_HS_IN_FS 
```

Arduino IDE (not tested)
```
-DUSBCON
-DUSBD_VID=0
-DUSBD_PID=0
-DUSBD_USE_CDC
-DHAL_PCD_MODULE_ENABLED
-DUSE_USB_HS
-DUSE_USB_HS_IN_FS
```
[image]
</details>

Both speed flags works as they should

valeros, 2020-08-11:
Thanks for the info. Could you please try the following config:
```
build_flags = 
  -DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC
  -DUSBCON
  -DUSBD_VID=0
  -DUSBD_PID=0
```
Thanks!

saloid, 2020-08-11:
Yep, it works

valeros, 2020-08-11:
And please check the last one option without VID, PID:
```
build_flags = 
  -DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC
  -DUSBCON
```

saloid, 2020-08-12:
Yes, it works even without VID PID flags, with just `-DPIO_FRAMEWORK_ARDUINO_ENABLE_CDC -DUSBCON`

11zombie11, 2020-08-15:
I have the same problem, usb serial not working with bluepill.

It compiles fine and the serial port shows on the PC but no data is sent.

It works fine with maple core.

Here’s the platformio.ini:

[env:STM32F103C8]
platform = ststm32
board = genericSTM32F103C8 ; 64K
; board = genericSTM32F103CB ; 128K
; board_build.core = maple
framework = arduino
upload_protocol = stlink
debug_tool = stlink

build_flags = 
  -ggdb
	-D USBCON
  -D USBD_USE_CDC

saloid, 2020-08-15:
Hi.
Replace `-D USBD_USE_CDC` with ' -DPIO_`RAMEWORK_ARDUINO_ENABLE_CDC`

11zombie11, 2020-08-15:
Also tried that. It basically does the same thing.

Raztou3D, 2021-05-21:
THANK YOU SO MUCH FOR THIS ! I was stuck with my Adafruit Feather STM32F405 Express board ... Now I can read the serial just fine !