platformio/platform-ststm32 #422: Upload protocol can't identify space char " " for dfu-util

thalesmaoa, 2020-08-11:
I found a bug but can't find a way to fix it.

```
[env:bluepill_f103c8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
upload_protocol = dfu
upload_port = /dev/ttyACM0
builf_flags = -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC -Os
```

My project folder:
```
$ pwd
/home/thales/Downloads/BlinkSTM32 Bootloader
```

When I try to upload:

```
$ pio run -v -t upload
Warning! Ignore unknown configuration option `builf_flags` in section [env:bluepill_f103c8]
Processing bluepill_f103c8 (platform: ststm32; board: genericSTM32F103C8; framework: arduino; upload_protocol: dfu; upload_port: /dev/ttyACM0; builf_flags: -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC -Os)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html
PLATFORM: ST STM32 8.0.0 > STM32F103C8 (20k RAM. 64k Flash)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200602 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.200628 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/bluepill_f103c8/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.1% (used 844 bytes from 20480 bytes)
Flash: [==        ]  15.3% (used 10020 bytes from 65536 bytes)
.pio/build/bluepill_f103c8/firmware.elf  :
section              size        addr
.isr_vector           268   134225920
.text                8796   134226188
.rodata              1088   134234984
.init_array            16   134236072
.fini_array             8   134236088
.data                 136   536870912
.bss                  708   536871048
.noinit                 0   536871756
._user_heap_stack    1540   536871756
.ARM.attributes        41           0
.comment              102           0
.debug_frame         1192           0
Total               13895
<lambda>(["upload"], [".pio/build/bluepill_f103c8/firmware.bin"])
AVAILABLE: blackmagic, dfu, jlink, serial, stlink
CURRENT: upload_protocol = dfu
MethodWrapper(["upload"], [".pio/build/bluepill_f103c8/firmware.bin"])
Use manually specified: /dev/ttyACM0
maple_upload ttyACM0 2 1EAF:0003 "/home/thales/Downloads/BlinkSTM32 Bootloader/.pio/build/bluepill_f103c8/firmware.bin"
Failed to open serial device.

dfu-util 0.7
/home/thales/Downloads/BlinkSTM32: No such file or directory
```

Now, if I remove the space from my folder name:

```
$ pwd
/home/thales/Downloads/BlinkSTM32Bootloader
```

And upload: 

```
$ pio run -v -t upload
Warning! Ignore unknown configuration option `builf_flags` in section [env:bluepill_f103c8]
Processing bluepill_f103c8 (platform: ststm32; board: genericSTM32F103C8; framework: arduino; upload_protocol: dfu; upload_port: /dev/ttyACM0; builf_flags: -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC -Os)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html
PLATFORM: ST STM32 8.0.0 > STM32F103C8 (20k RAM. 64k Flash)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200602 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.200628 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/bluepill_f103c8/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.1% (used 844 bytes from 20480 bytes)
Flash: [==        ]  15.3% (used 10020 bytes from 65536 bytes)
.pio/build/bluepill_f103c8/firmware.elf  :
section              size        addr
.isr_vector           268   134225920
.text                8796   134226188
.rodata              1088   134234984
.init_array            16   134236072
.fini_array             8   134236088
.data                 136   536870912
.bss                  708   536871048
.noinit                 0   536871756
._user_heap_stack    1540   536871756
.ARM.attributes        41           0
.comment              102           0
.debug_frame         1192           0
Total               13895
<lambda>(["upload"], [".pio/build/bluepill_f103c8/firmware.bin"])
AVAILABLE: blackmagic, dfu, jlink, serial, stlink
CURRENT: upload_protocol = dfu
MethodWrapper(["upload"], [".pio/build/bluepill_f103c8/firmware.bin"])
Use manually specified: /dev/ttyACM0
maple_upload ttyACM0 2 1EAF:0003 "/home/thales/Downloads/BlinkSTM32Bootloader/.pio/build/bluepill_f103c8/firmware.bin"
Failed to open serial device.
No valid DFU suffix signature
Warning: File has no DFU suffix

dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x1eaf product = 0x0003
Opening DFU capable USB device... ID 1eaf:0003
Run-time device DFU version 0110
Found DFU: [1eaf:0003] devnum=0, cfg=1, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
bytes_per_hash=206
Copying data from PC to DFU device
Starting download: [##################################################] finished!
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode
Waiting for /dev/ttyACM0 serial...Done
======================================================================================= [SUCCESS] Took 8.98 seconds =======================================================================================
```

thalesmaoa, 2020-08-12:
I found the problem.
```
~/.platformio/packages/tool-stm32duino$ nano maple_upload
```
at line 40:
```
"${DFU_UTIL}" -d ${usbID} -a ${altID} -D ${binfile} ${dfuse_addr} -R
```
double quote mark is missing
```
"${DFU_UTIL}" -d ${usbID} -a ${altID} -D "${binfile}" ${dfuse_addr} -R
```

Before you consider that this is an STM32duino bug, I've checked.
https://github.com/stm32duino/Arduino_Tools/blob/master/linux/maple_upload.sh
Not only the file is different, but it is correct:

```
"${DIR}/dfu-util.sh" -d "${usbID}" -a "${altID}" -D "${binfile}" ${dfuse_addr} -R
```

I'm not sure where platformio push ststm32 files, but it is outdated.

maxgerhardt, 2021-03-14:
@valeros This bug is still unfixed and people are hitting this bug in e.g. https://community.platformio.org/t/platformio-newbie-hitting-wall-after-wall/19971