platformio/platform-ststm32 #424: Errors importing Serial

waymond91, 2020-08-17:
### Description of problem
[x] **PlatformIO Core**.

Hello, I use PlatformIO to compile & update code for a remote servo system on an STM32F411-based system. Everything was working fine, but while working on a new update something broke. So I reverted the project to the last stable release and I am still getting the same compiler error.

With the reverted build, platformio fails with:
`ImportError: cannot import name 'Serial' from 'serial' `
So then I downgraded to 4.3.3 and it failed with 
`'Serial' does not name a type; did you mean 'serial_t'?`

Anyway, hopefully I haven't made a big goof, but I'm not sure how to proceed.
I believe that it broke on the remote machine as well, but I will double check.
### Configuration

**Operating system**:
Linux g7 5.4.0-42-generic platformio/platformio-core#46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

**PlatformIO Version** (`platformio --version`):
4.3.4 & 4.3.3

```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:nucleo_f411re]
platform = ststm32
board = nucleo_f411re
framework = mbed
debug_tool = stlink
upload_protocol = stlink
build_flags = -ggdb -g3 
```

**Source file to reproduce issue:**
```cpp
#include "mbed.h"
#include "QEI.h"
//#include "../lib/UART_ISR.h"
#include "../lib/PinDefs.h"
#include "../lib/MotorTests.h"
#include "../lib/PID.h"

void serial_update(void);

void control_loop(void);
Ticker ticker;
void mot1_set(float);
void mot2_set(float);

Serial pc(PA_11, PA_12); //tx, rx;
PID pid1(0.0175, 0.0, 0.1, &wheel1, &ticker);
PID pid2(0.0134, 0.0, 0.1, &wheel2, &ticker);

int main(){

  pc.baud(PC_BAUD);

  buzzer.period(0.0015);
  buzzer = 0.5;
  wait_us(500000);
  buzzer = 0;

  m1in1 = 0;
  m1in2 = 1;
  m1d1 = 1;
  m1d1.period(.00005);

  m2in1 = 0;
  m2in2 = 1;
  m2d1 = 1;
  m2d1.period(.00005);

  //pc.attach(&rxIsr);
  ticker.attach_us(&control_loop, 5000.0);

  while(1) {
    serial_update();
  }
}

void control_loop(void){

  mot1_set(pid1.update());
  mot2_set(pid2.update());

}

// Updates PID setpoints from new serial input
void serial_update(void){
	//Only update if there is new line in the serial buffer
	//t.reset();
	if(newlineFlag == 1){
		string local;
		local = in1;
		newlineFlag = 0;

		//Check to see which wheel to update
		if((local.find(left) == 0) & (local.find(end_char) == local.length()-1)){
			local.erase(local.begin());
			local.erase(local.end()-1);
			int dl;
			dl = std::atoi(local.c_str());
			pid2.setpoint += dl;

      pc.printf("L:%i\r\n", dl);

		}

		if((local.find(right) == 0) & (local.find(end_char) == local.length()-1)){
			local.erase(local.begin());
			local.erase(local.end()-1);
			int dr;
			dr = std::atoi(local.c_str());
			pid1.setpoint -= dr;

      pc.printf("R:%i\r\n", dr);
		}

    if((local.find(test_command) == 0) & (local.find(end_char) == local.length()-1)){
      test_t pid_test;
      local.erase(local.begin());
      local.erase(local.end()-1);
      const char * c_str = local.c_str();
      int side;
      sscanf(c_str, "%d,%f,%f,%f,%f", &side, &pid_test.p, &pid_test.i, &pid_test.d, &pid_test.setpoint);

      if (side == 1){
        pid1.tune(pid_test);
      }else if(side == 2){
        pid2.tune(pid_test);
      }

    }
  }
}

```

### Compile Errors
Results with the newest version of platformio:
```
Processing nucleo_f411re (platform: ststm32; board: nucleo_f411re; framework: mbed)
-----------------------------------------------------------------------------------------------------
ImportError: cannot import name 'Serial' from 'serial' (/home/brett/anaconda3/lib/python3.7/site-packages/serial/__init__.py):
  File "/home/brett/anaconda3/lib/python3.7/site-packages/platformio/builder/main.py", line 89:
    env = DefaultEnvironment(**DEFAULT_ENV_OPTIONS)
  File "/home/brett/.platformio/packages/tool-scons/script/../engine/SCons/Defaults.py", line 88:
    _default_env = SCons.Environment.Environment(*args, **kw)
  File "/home/brett/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 990:
    apply_tools(self, tools, toolpath)
  File "/home/brett/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 102:
    env.Tool(tool)
  File "/home/brett/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 1809:
    tool = SCons.Tool.Tool(tool, toolpath, **kw)
  File "/home/brett/.platformio/packages/tool-scons/script/../engine/SCons/Tool/__init__.py", line 122:
    module = self._tool_module()
  File "/home/brett/.platformio/packages/tool-scons/script/../engine/SCons/Tool/__init__.py", line 239:
    module = spec.loader.load_module(spec.name)
  File "<frozen importlib._bootstrap_external>", line 407:
    
  File "<frozen importlib._bootstrap_external>", line 907:
    
  File "<frozen importlib._bootstrap_external>", line 732:
    
  File "<frozen importlib._bootstrap>", line 265:
    
  File "<frozen importlib._bootstrap>", line 696:
    
  File "<frozen importlib._bootstrap>", line 677:
    
  File "<frozen importlib._bootstrap_external>", line 728:
    
  File "<frozen importlib._bootstrap>", line 219:
    
  File "/home/brett/anaconda3/lib/python3.7/site-packages/platformio/builder/tools/pioupload.py", line 26:
    from serial import Serial, SerialException
```
So then I tried to rollback to version 4.3.3 and got a different error

```
***********************************************************************************
Obsolete PIO Core v4.3.3 is used (previous was 4.3.4)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
***********************************************************************************
Processing nucleo_f411re (platform: ststm32; board: nucleo_f411re; framework: mbed)
-----------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f411re.html
PLATFORM: ST STM32 8.0.0 > ST Nucleo F411RE
HARDWARE: STM32F411RET6 100MHz, 128KB RAM, 512KB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES: 
 - framework-mbed 6.60200.200722 (6.2.0) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
...
src/main.cpp: At global scope:
src/main.cpp:15:1: error: 'Serial' does not name a type; did you mean 'serial_t'?

```

waymond91, 2020-08-17:
Ok so I created a new, empty project just to make sure that it wasn't something horrible I did in my code.
And to make the issue easier to reproduce:
### main.cpp
```cpp
#include <mbed.h>
Serial pc(PA_11, PA_12); //tx, rx;

int main() {

  // put your setup code here, to run once:

  while(1) {
    // put your main code here, to run repeatedly:
  }
}

```
### platformio.ini
```ini
[env:nucleo_f411re]
platform = ststm32
board = nucleo_f411re
framework = mbed
```

### PlatformIO Version:
```
$ pio --version
PlatformIO, version 4.3.4
```

### Operating System
```
Distributor ID:	Ubuntu
Description: Ubuntu 20.04 LTS
Release:	20.04
Codename: focal
```

### Tool Chain
```
$ python --version
Python 3.7.4
$ which python
/home/brett/anaconda3/bin/python
$ which pio
/home/brett/anaconda3/bin/pio
```

valeros, 2020-08-17:
I see here two problems:

> 'Serial' does not name a type; did you mean 'serial_t'?

It looks like your code is not compatible with the latest Mbed OS v6. Please read this [release note](https://community.platformio.org/t/support-for-mbed-os-6-stable-and-mature-apis-cloud-services-support-enhancements-to-the-bare-metal-profile/15079) about the recent changes in Mbed ecosystem.

> ImportError: cannot import name 'Serial' from 'serial'

As for that problem, I'd recommend you not to waste your time fixing that import error and create a new venv and install PlatformIO from scratch.

waymond91, 2020-08-17:
Thank you for the help! 
The link you posted did the trick.
I'm not interest in MBEDs RTOS for this project, so for posterity my platformio.ini looks like:

```ini
[env:nucleo_f411re]
platform = ststm32
board = nucleo_f411re
framework = mbed
debug_tool = stlink
upload_protocol = stlink
build_flags = -ggdb -g3
platform_packages =
    framework-mbed @ ~6.51504.0
```

closed 2020-08-17 by waymond91