platformio/platform-ststm32 #428: No module named 'build_api' error

xtrinch, 2020-08-29:
Trying to create a new project for disco f769ni board with mbed, but running into the following error upon building: 

```
> Executing task: C:\Users\xtrinch\.platformio\penv\Scripts\platformio.exe run <

Processing disco_f769ni (platform: ststm32; board: disco_f769ni; framework: mbed)
------------------------------------------------------------------------------------------------------------------------------------- 
PackageManager: Installing toolchain-gccarmnoneeabi @ ~1.90201.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
toolchain-gccarmnoneeabi @ 1.90201.191206 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_f769ni.html
PLATFORM: ST STM32 8.0.0 > ST 32F769IDISCOVERY
HARDWARE: STM32F769NIH6 216MHz, 512KB RAM, 1MB Flash
DEBUG: Current (stlink) On-board (stlink) External (blackmagic, jlink)
PACKAGES:
 - framework-mbed 6.60200.200722 (6.2.0)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
ImportError: No module named build_api:
  File "C:\Users\xtrinch\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\xtrinch\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\xtrinch\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\xtrinch\.platformio\platforms\ststm32\builder\main.py", line 99:
    target_elf = env.BuildProgram()
  File "C:\Users\xtrinch\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\xtrinch\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\xtrinch\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\xtrinch\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 118:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\xtrinch\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\xtrinch\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 330:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\xtrinch\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 604:
    return method(*args, **kw)
  File "C:\Users\xtrinch\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\xtrinch\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\xtrinch\.platformio\platforms\ststm32\builder\frameworks\mbed.py", line 37:
    "platformio-build.py"))
  File "C:\Users\xtrinch\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 604:
    return method(*args, **kw)
  File "C:\Users\xtrinch\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\xtrinch\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\xtrinch\.platformio\packages\framework-mbed\platformio\platformio-build.py", line 43:
    from pio_mbed_adapter import PlatformioMbedAdapter
  File "C:\Users\xtrinch\.platformio\packages\framework-mbed\platformio\pio_mbed_adapter.py", line 23:
    from tools.build_api import prepare_toolchain, UPDATE_WHITELIST
==================================================== [FAILED] Took 96.09 seconds ==================================================== 
```

I'm on windows 10 64bit, running 1.48.2 VSCode.

Any ideas?

valeros, 2020-08-31:
Hi @xtrinch ! Looks like you mbed package is broken in some way. Could you please try to delete `C:\Users\xtrinch\.platformio\packages\framework-mbed` folder and run your example again?

xtrinch, 2020-09-01:
Hi @valeros,

Followed your instructions, deleted the folder, tried again with no success:
```
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_smartcard_ex.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_smbus.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_spdifrx.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_spi.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_spi_ex.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_sram.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_tim.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_tim_ex.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_uart.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_uart_ex.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_usart.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_hal_wwdg.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_adc.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_crc.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_dac.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_dma.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_dma2d.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_exti.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_fmc.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_gpio.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_i2c.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_lptim.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_pwr.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_rcc.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_rng.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_rtc.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_sdmmc.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_spi.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_tim.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_usart.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_usb.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\STM32F7xx_HAL_Driver\stm32f7xx_ll_utils.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\STM32Cube_FW\system_stm32f7xx.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\TARGET_STM32F769xI\TARGET_DISCO_F769NI\PeripheralPins.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\TARGET_STM32F769xI\TARGET_DISCO_F769NI\system_clock.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\TARGET_STM32F769xI\TOOLCHAIN_GCC_ARM\startup_stm32f769xx.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\analogin_device.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\analogout_device.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\flash_api.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\gpio_irq_device.o
Compiling .pio\build\disco_f769ni\FrameworkMbed\targets\TARGET_STM\TARGET_STM32F7\pwmout_device.o
[... cut ...]
C:\Users\xtrinch\.platformio\packages\framework-mbed\rtos\source\TARGET_CORTEX\TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:77: undefined reference to `main'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\disco_f769ni\firmware.elf] Error 1
=================================================== [FAILED] Took 974.87 seconds ===================================================
The terminal process "C:\Users\xtrinch\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
```

valeros, 2020-09-02:
Looks like the framework is compiled fine, but the linker couldn't find the `main` function. What is the content of your `main.cpp`?

xtrinch, 2020-09-02:
Oh well I never got around to putting anything in there because I was stopped by the first error, so the second one's on me. Compiles fine with a main.cpp 👍 
Thanks for your help!

closed 2020-09-02 by valeros