platformio/platform-ststm32 #435: How to use "attach Interrupt" ?

Voha888, 2020-09-24:
Chip is F103C8T6 (Blue pill)
I want to count the number of pulses per second. 
My scetch is:

```
#define FLOW_SENSOR_PIN PB3

volatile uint32_t flow_counter=0;

void flow_sensor_ISR(){

flow_counter++;

}

void flow_sensor_setup(){

attachInterrupt(digitalPinToInterrupt(FLOW_SENSOR_PIN), flow_sensor_ISR, CHANGE);

}
```

But **flow_counter** every time is 0.

I tried **bluepill_f103c8**, **genericSTM32F103C8** and **blackpill_f103c8** platforms

fpistm, 2020-09-25:
You have ton configure the pin in input, for example:
`pinMode(interruptPin, INPUT_PULLUP;`

https://www.arduino.cc/reference/en/language/functions/external-interrupts/attachinterrupt/?setlang=it

valeros, 2020-09-25:
Closing the issue, as this repository is not the right place to ask generic questions about programming.

closed 2020-09-25 by valeros

Voha888, 2020-09-25:
> pinMode(interruptPin, INPUT_PULLUP;

Thank You! But its **not working**,
i trying change pin for B3 as:

`#define FLOW_SENSOR_PIN PB3`
or
`#define FLOW_SENSOR_PIN D3`

for A0 as:
`#define FLOW_SENSOR_PIN PA0`
or
`#define FLOW_SENSOR_PIN A0`

But flow_counter every time is 0.

fpistm, 2020-09-25:
@Voha888 
Don't know what you made exactly. I've tested and it works with the following code:

```
#define FLOW_SENSOR_PIN PB3

volatile uint32_t flow_counter = 0;

void flow_sensor_ISR() {
  flow_counter++;
}

void setup() {
  Serial.begin(9600);
  pinMode(FLOW_SENSOR_PIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(FLOW_SENSOR_PIN), flow_sensor_ISR, CHANGE);
}

void loop(){
  Serial.printf("flow_counter is %i\n", flow_counter);
  delay(1000);
}
```

Note that I've tested with Arduino IDE not PIO.

I don't understand you point with Hw I2C. Anyway the pins used for interrupt should not be used for other. 1 function per pin.

As @valeros mentioned above, if you need more support on this please use the forum: https://www.stm32duino.com

Voha888, 2020-09-25:
> Note that I've tested with Arduino IDE not PIO.

Yes, I also ran this code in Arduino and it works right there!

But it doesn't work in PIO , @valeros  i think, its issue? or no?

fpistm, 2020-09-25:
Try adding `extern "C"` before the ISR prototype just for testing with PIO.

Voha888, 2020-09-25:
> Try adding `extern "C"` before the ISR prototype just for testing with PIO.

You are right, these are code issues. I created a sketch in PIO in one file, without the project structure, and the interrupts began to work. 

I will try your advice, as well as try to understand what this may be due to. Sorry for the false alarm, you helped me a lot, thank you very much!