platformio/platform-ststm32 #439: HAL drivers are out of date

dingyifei, 2020-10-07:
I found framework-stm32cube\f4\Drivers\STM32F4xx_HAL_Driver haven't get updated for a year. This leads to some incompatability with code generated by STM32CubeMx using the newest framework from https://github.com/STMicroelectronics/STM32CubeF4. For example, htim2.Init.AutoReloadPreload is not defined in the old framework package and had to comment out manually. 

I tried to manually include the newest driver but it wasn't working. 
Here are the edits I made to my ini file:
```
build_flags =
  +  -I Drivers/STM32F4xx_HAL_Driver/Inc
lib_extra_dirs = 
 +   Drivers/STM32F4xx_HAL_Driver
```
They didn't work and I believe it was firmware package related because I got the following error (I only copied a section of it):
"C:\Users\13676\.platformio\packages\framework-stm32cube\f4\Drivers\STM32F4xx_HAL_Driver\Src\stm32f4xx_hal_pcd.c: In function 'HAL_PCD_ActivateRemoteWakeup':
Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_ll_usb.h:397:52: error: 'USBx_BASE' undeclared (first use in this function);
did you mean 'SCB_BASE'?
 #define USBx_DEVICE     ((USB_OTG_DeviceTypeDef *)(USBx_BASE + USB_OTG_DEVICE_BASE))"

Is there any way to partially replace `FRAMEWORK_DIR` in the python script `stm32cube.py` or update the framework package?

gmcn42, 2020-10-08:
I am running into the same problem with a Nucleo STM32H743 board. Current stable framework version in CubeMX is 1.8.0 from May 29, 2020. In platformio, newest update is 1.3.0 from July 4, 2018. So more than 2 years and 5 releases old. Right now, a bare minimum project for the Nucleo board generated in CubeMX will not compile due to unimplemented functions and undeclared constants.

Some error messages from the failed compilation:

```
src/main.c:276:15: error: 'UART_InitTypeDef {aka struct <anonymous>}' has no member named 'ClockPrescaler'; did you mean 'Prescaler'?
   huart3.Init.ClockPrescaler = UART_PRESCALER_DIV1;
src/system_stm32h7xx.c:153:6: error: 'FLASH_LATENCY_DEFAULT' undeclared (first use in this function); did you mean 'FLASH_LATENCY_1'?
   if(FLASH_LATENCY_DEFAULT  > (READ_BIT((FLASH->ACR), FLASH_ACR_LATENCY)))
      ^~~~~~~~~~~~~~~~~~~~~
      FLASH_LATENCY_1
src/system_stm32h7xx.c:153:6: note: each undeclared identifier is reported only once for each function it appears in
src/system_stm32h7xx.c:261:3: error: 'FMC_Bank1_R' undeclared (first use in this function); did you mean 'FMC_Bank5_6'?
   FMC_Bank1_R->BTCR[0] = 0x000030D2;
   ^~~~~~~~~~~
   FMC_Bank5_6
*** [.pio/build/nucleo_h743zi/src/system_stm32h7xx.o] Error 1
```

....et cetera. Seems it's high time, the libraries get updated.

At
[https://community.platformio.org/t/it-looks-like-package-framework-stm32cube-f4-is-outdated/12886/6] a user posted a workaround involving a self-made repository to let platformio pull in the newer sources from there but this should really be fixed 'officially'.

dingyifei, 2020-10-08:
> I am running into the same problem with a Nucleo STM32H743 board. Current stable framework version in CubeMX is 1.8.0 from May 29, 2020. In platformio, newest update is 1.3.0 from July 4, 2018. So [...]

If anyone in platformio dev team would publish the package on Github it would be awesome. The stm32cube fw is released on Git and making it a submodule of the fw package should be easy.
 https://github.com/STMicroelectronics/STM32CubeF4/releases

dingyifei, 2020-10-08:
I made some rough changes that might solve the problem, not sure I correctly implemented them since I didn't read main.py...
NOTE: I haven't test it..... (I'll test it after I get free time or something xD) Anyone want to test it?
https://github.com/platformio/platform-ststm32/compare/develop...dingyifei:develop

GunZi200, 2020-10-21:
Just took a look myself and wow everything is out of date. 

| Package           | PIO version | Newest version | Release notes preview |
|----------------|---------------------|------------------|-----------------------|
| F0                   | 1.9.0                        | 1.11.1              |  [Github Preview](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeF0/blob/master/Release_Notes.html)  |
| F1                   | 1.7.0                        | 1.8.2                 |  [Github Preview](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeF1/blob/master/Release_Notes.html)  |
| F2                   | 1.7.0                        | 1.9.1                 |  [Github Preview](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeF2/blob/master/Release_Notes.html) |
| F3                   | 1.10.0                        | 1.11.1                 |  [Github Preview](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeF3/blob/master/Release_Notes.html) |
| F4                   | 1.23.0                       | 1.25.1                 |  [Github Preview](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Release_Notes.html) |
| F7                   | 1.14.0                        | 1.16.0                |  [Github Preview](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeF7/blob/master/Release_Notes.html)  |
| H7                   | 1.3.0                        | 1.8.0                 |  [Github Preview](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Release_Notes.html)  |
| L0                   | 1.11.0                        | 1.11.3                 |  [Github Preview](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeL0/blob/master/Release_Notes.html)  |
| L1                   | 1.8.1                        | 1.10.1                 |  [Github Preview](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Release_Notes.html) |
| L4                   | 1.13.0                        | 1.16.0                 |  [Github Preview](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeL4/blob/master/Release_Notes.html)  |

I noticed when going through these versions that `framework-stm32cube` does not include STM32L5, STM32WB, STM32G4, STM32G0

maxgerhardt, 2020-10-21:
Yeah these versions are ancient. I'll just cut it short and generate a new package version of `framework-stm32cube`. The same issue also goes for the `framework-spl` package, but it's less used so this doesn't pop up as often. I'll then hope for quick integration by @valeros

maxgerhardt, 2020-10-22:
I've created a script which can automatically download the latest versions of STM32Cube packages for every available series and create a `framework-stm32cube` package from it: https://github.com/maxgerhardt/pio-stm32cube-package-creator 

I'm currently testing the created packet. Compilation of the current STM32Cube-HAL-Blink example goes through for a F0 target at least. Now on to the others..

GunZi200, 2020-10-23:
Is there anything else we can do to help them update this? 

Maybe we should create a separate issue for the G0, G4, L5, MP1 and WB?

arrowcircle, 2020-11-19:
Hey! Any updates on this?

maxgerhardt, 2020-11-19:
I wasn't really able to find time to do extensive testing of my framework package, so I'm appreciating support and testing from others regarding all the example projects and here more STM32CubeMX generated projects with like an RTOS etc. Also hoping for integration by @valeros  :)

valeros, 2020-11-26:
Resolved in https://github.com/platformio/platform-ststm32/commit/ec29fab5f4e75ad7664cefcfc723e3acc6e22592. Sorry for such a late update.

closed 2020-11-26 by valeros

arrowcircle, 2020-11-27:
@valeros Thanks!
Is HAL updated to latest versions? When this will be released and available for use?

maxgerhardt, 2020-11-27:
Since it's in the dev branch you should be able to use it in a PlatformIO project just by replacing the `platform = ` line in the `platformio.ini` to 

```
platform = https://github.com/platformio/platform-ststm32.git
```

insolace, 2024-12-28:
Is there any plan to solve/update this? The platformio STM32G0 drivers are almost 4 years old and missing some key bug fixes.