platformio/platform-ststm32 #443: Support Kotlin Native with Zephyr

ajsb85, 2020-10-25:
**Support Kotlin Native with Zephyr**

URL for references:
https://kotlinlang.org/docs/tutorials/native/using-command-line-compiler.html
https://github.com/JetBrains/kotlin-native/blob/master/samples/zephyr/src/main.kt

Kotlin native support from several years ago the STM32 board as first example:

- Support first embedded target (STM32 board)
- STM32 blinky demo in pure Kotlin/Native

ajsb85, 2020-10-27:
Hi, @valeros 
I am building a big community around it.
How we can help us to push rolling out it?

valeros, 2020-10-30:
Hi @ajsb85 ! Unfortunately I have no experience with Kotlin, so the best to way to push it forward is to provide a PR with the initial implementation.

ajsb85, 2020-10-30:
But guide me to know where I need to fill out.
It means in the future we will use Kotlin instead C for all the boards.

ajsb85, 2020-10-30:
I am building a big community around it from my own company.
And we want to release a development kit in India in the coming months.

ajsb85, 2020-10-30:
We will have a marketplace of MikroBUS and Kotlin libraries for it.

valeros, 2020-10-30:
I have no idea of the build workflow used in Kotlin, but here is the main script for `ststm32` platform https://github.com/platformio/platform-ststm32/blob/develop/builder/main.py that is responsible for setting all major tools used for compilation.

closed 2025-07-23 by valeros