platformio/platform-ststm32 #446: Ststm32 maple core outdated!

CodeOhms, 2020-11-03:
The ststm32 [maple core is outdated](https://github.com/arpruss/USBComposite_stm32f1/issues/55)! Please update it so users don't have to do it manually - especially as doing it manually is not officially supported.

As a result I cannot compile libraries such as [USBComposite for STM32F1](https://github.com/arpruss/USBComposite_stm32f1)!

sjasonsmith, 2020-11-29:
This would be very helpful for Marlin users, since newer Maple versions produce substantially smaller builds. I'm seeing about a 30k size reduction in compiled firmware.

I realize that this isn't the preferred framework anymore, and we are gradually working our way off of it, but for now a lot of users still depend on it.

closed 2020-11-30 by valeros

sjasonsmith, 2020-11-30:
@valeros thank you so much for this! We thought we would be content with the old Maple until we finished switching frameworks, until somebody pointed out the binary size advantage in newer versions.

commit 30ee7d12 (2020-11-30): Update Arduino maple core // Resolve #446