platformio/platform-ststm32 #449: Enable better IntelliSense in STM32Cube framework by adding Src path to search path

maxgerhardt, 2020-11-25:
As explained in https://community.platformio.org/t/a-detailed-description-of-the-functions-in-the-platformio/17369/5. 

When you create a STM32Cube project in VSCode, e.g. for a board of the L4 series, you don't get the full intended autocomplete, that is, function descriptions, parameter descriptions etc are missing. This does work perfectly in fine in the competitor CubeIDE.

The source of the problem is that the STM32HAL has the actually detailed Doyxgen documentation in the `Src/` folder of the HAL, not in the `Inc/` folder, which is currently that what PlatformIO includes in the search path. As can be seen in the topic, after manually adding `../l4/Drivers/STM32L4xx_HAL_Driver/Src` in the `c_cpp_properties.json`, the IntelliSense goes from

[image]

to

[image]

This should be easily fixable by adding the `Src/` path too in the `stm32cube.py` builder script.

maxgerhardt, 2020-11-26:
Oh wow that was a **massive** update, also all the STM32Cube packets were split up now according to their STM series? The underlying STM32HAL package was updated as well, thus resolving https://github.com/platformio/platform-ststm32/issues/439?

valeros, 2020-11-26:
You're right, we now use the official packages. I also added some tweaks to configure the framework in a bit more flexible manner.