platformio/platform-ststm32 #462: Add documentation for board_build.variants_dir option

maxgerhardt, 2020-12-28:
The use case as stated in https://community.platformio.org/t/how-use-a-same-chip-but-with-different-environment/18082 is pretty valid. Instead of the framework's variant folder, like https://github.com/stm32duino/Arduino_Core_STM32/tree/master/variants, some projects require a custom variant folder location and variant. 

This is possible since the builder scripts allows configuration of that 

https://github.com/stm32duino/Arduino_Core_STM32/blob/b626ed2142b9dac6147d44336f417ad91e091879/tools/platformio-build.py#L45-L49

and the board allows reconfiguration of the variant by a change of `board_build.variant = ...` in the `platformio.ini`.

However, documentation for that is lacking in https://docs.platformio.org/en/latest/platforms/ststm32.html#configuration or the board page https://docs.platformio.org/en/latest/boards/ststm32/nucleo_l476rg.html. I think it should be added (and this feature should be maintained / preserved over time) in the ST STM32 configuration page.

pat1, 2021-08-30:
in st32duino version >= 2 variant definitions was changed :
https://github.com/stm32duino/Arduino_Core_STM32/releases/tag/2.0.0
https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-%28board%29

you can find new example here:
https://github.com/r-map/rmap/blob/master/platformio/test/microduinostm32_F103CB_i2cscanner/platformio.ini

The problem not covered by board_build.variants_dir option is that is not possible to define a new board in a exportable way inside a project but inside  core_dir   as documented at https://docs.platformio.org/en/latest/platforms/creating_board.html
Will be useful to have some board_build.<options> to define a path for locally defined board

maxgerhardt, 2021-09-27:
This issue is still not resolved. @valeros can you transfer it to https://github.com/platformio/platformio-docs/issues?